mbuesch/razer #39: razercfg vs qrazercfg

ptrj, 2015-01-16:
whit qrazercfg i can set every resolution, like in Res5 2500DPI:

[image]

but when i try razercfg: 
ptrj@kuralin:~$ razercfg -r 5:2500
Invalid resolution 2500
ptrj@kuralin:~$ razercfg -r 5:1000
ptrj@kuralin:~$ razercfg -r 5:1800
ptrj@kuralin:~$ razercfg -r 5:2500
Invalid resolution 2500

it doesnt work :-(

mbuesch, 2015-01-16:
Please try something like this:

$ razercfg -r 5:1000x1000
or
$ razercfg -r 5:3

This should both work and result in 1000 DPI x 1000 DPI

ptrj, 2015-01-17:
ptrj@kuralin:~$ razercfg -r 5:1000x1000
Invalid resolution 1000
ptrj@kuralin:~$ razercfg -r 5:3
ptrj@kuralin:~$ razercfg -R
Profile  1*:    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile  2 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile  3 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile  4 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile  5 :    1 (800/800 DPI),  2 (1800/1800 DPI), *3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile  6 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile  7 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile  8 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile  9 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile 10 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile 11 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile 12 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile 13 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile 14 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile 15 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile 16 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile 17 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile 18 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile 19 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)
Profile 20 :    1 (800/800 DPI), *2 (1800/1800 DPI),  3 (3500/3500 DPI),  4 (5600/5600 DPI),  5 (10000/10000 DPI)

nothing, profil 5 is on 10000DPI

ptrj, 2015-01-17:
btw: led colors works perfect :-)

mbuesch, 2015-01-17:
1000x1000 is not a valid request for your configuration.
And the -r 5:3 request seems to have worked just fine. Profile 5 is set to resolution index 3 (which is 3500x3500).

ptrj, 2015-01-17:
is it posibble to set this five resolution colomn permanently ?

mbuesch, 2015-01-17:
I think for this device there currently is no way to do so.

ptrj, 2015-01-17:
i allready understand how parameter "-r" das work :-) i miss parameter for set this five "resolution DPI" column wich ar in qrazercfg and i can change it. Problem is when i reset my PC, mouse is reseted to default. And i muss open qrazercfg and setup mouse again :-(

mbuesch, 2015-01-17:
Yes, I understand.
There currently is no way to do that.

I will take patches or pulls adding such a command to razercfg.

ptrj, 2015-01-17:
much thanks

closed 2015-01-17 by ptrj