mbuesch/razer #40: Can't build 0.31

masterpat, 2015-01-23:
When running the "make" Command I get the following errors:

[  5%] Building C object librazer/CMakeFiles/razer.dir/hw_deathadder_chroma.o
/home/patrick/Dokumente/razer/librazer/hw_deathadder_chroma.c:146:15: error: expected declaration specifiers or ‘...’ before ‘sizeof’
/home/patrick/Dokumente/razer/librazer/hw_deathadder_chroma.c:147:8: error: expected declaration specifiers or ‘...’ before string constant
make[2]: *** [librazer/CMakeFiles/razer.dir/hw_deathadder_chroma.o] Error 1
make[1]: *** [librazer/CMakeFiles/razer.dir/all] Error 2
make: *** [all] Error 2

I'm using Debian 7.80 GNOME and razercfg 0.31.
razercfg 0.30 was working.

mbuesch, 2015-01-24:
Thanks for your bug report.

What compiler version are you using?
Please try to remove lines 146, 147 and 148 of librazer/hw_deathadder_chroma.c and try to compile.

masterpat, 2015-01-24:
I use the preinstalled GNU g++-4.7.

And it worked by removing these lines, thanks. :)

mbuesch, 2015-01-24:
Thanks.
This should be fixed by f6512a514a213e2781623b3433dd968eef3565e1

closed 2015-01-24 by mbuesch