mbuesch/razer #43: Support for Orochi?

spacepluk, 2015-03-31:
Any chance we get support for this model?

Thanks.

mbuesch, 2015-04-01:
Nobody is currently working on it.

spacepluk, 2015-04-01:
how hard would it be to add it?  It seems like a simple mouse... is there anything I can read to get a feel of what needs to be done?

mbuesch, 2015-04-01:
Here are a few words on adding support for new hardware:
http://bues.ch/cms/hacking/razercfg.html#how_to_add_support_for_new_devices

I'm happy to merge any properly developed drivers.
Just keep in mind that there is no such thing as a simple Razer mouse :)

Also see this, if you want to contribute:
http://bues.ch/cms/resources/contribute.html

spacepluk, 2015-04-01:
hehehe, thanks!
I'll put it on my list of fun stuff I want to do :)

mbuesch, 2016-01-03:
Is this still on the list? :)

spacepluk, 2016-01-03:
haha, it's still on the list but it keeps growing! :)

spacepluk, 2017-01-01:
Not from me, it's gonna take a while until I'm able to put some time into this.

z3ntu, 2017-01-01:
That mouse is supported by [razer-drivers](https://github.com/terrycain/razer-drivers), so you can use that (at least) until razercfg gets support.

z3ntu, 2017-01-01:
Open an issue in the razer-drivers repo about the device and terry will implement it!