mbuesch/razer #47: Macros for Naga Hex

Zeioth, 2015-08-16:
I readed that there's no plan to implement macros for razer naga hex because they are stored in the cloud. So, there's any alternative for gaming macros in linux? I tried xbindkeys but it's no suitable for gaming because it has a lot of delay and bugs. Thank you.

mbuesch, 2015-08-17:
If there are bugs in xbindkeys, please report that to the xorg project.

razercfg can't and won't implement such functionality. razercfg is supposed to configure the hardware. It does not interact with X11 or the kernel input driver. But remapping must happen in X11 or the kernel driver.

OvermindDL1, 2015-08-17:
The Razer Naga side keys are generic keyboard keys.  I remap mine (I have the 2014 12 key model) to the F13 to F24 keys for ease of non-conflicting use, then use a set of global listeners to perform command on their press.  Works well and easy to setup.  Message me if you want any of my code.  The only thing I cannot get working right is the tilt scrolling because of an evdev bug and I do not have the time to write a filter for it...

closed 2015-09-05 by mbuesch