mbuesch/razer #49: Adding save to razer.conf from qrazercfg

rcmn, 2015-09-04:
On linux using qrazercfg with or without administrative privilege does not save the configuration.Making qrazercfg a bit useless. As a result the configuration of the device reset to factory default after being unplugged or the system is rebooted. Manually configuring razer.conf allow for a more permanent configuration.

mbuesch, 2015-09-05:
I would not call that "making qrazercfg useless", but anyway.

I think that functionality has to be implemented in razerd.
The plan basically was to have a second razer.conf (somewhere in /var or such) that is written by razerd to reflect the user set configuration. On startup razerd would take that file and reload the settings, just like it does with the static etc file now. (Code can be shared).
I'm not sure yet, whether the etc should override the var config settings or vice versa. In any case it should happen on a per-setting and not on a per-file basis.

rcmn, 2015-09-07:
Sorry Michael for the poor choice of word.
Because razed,razercfg and qrazercfg work so well and accomplish so well
their goals , that not having this arguably essential option for the end
user (and easy to implement option in comparison to what you have already
accomplished) that it seemed to stand out.

Anyway, what is the use of razer.conf in etc ? Because at the moment
/etc/razer.conf is created by the user so it is the "user set
configuration". What would accomplish to have 2 conf files ? .

Now What I see in other program is to have a default config file
(/etc/razer.conf) and the user settings are saved in the user
space(ex:/home/.razer/razer.conf or something).

mbuesch, 2015-09-07:
/etc/razer.conf is supposed to be some kind of default configuration set. It is a sane configuration that most users should be able to work with somehow. Contrary to Razer's default settings of some mice.

As you can see in the config format there is some simple support for wildcards in the section identifiers. And the /etc/razer.conf uses these wildcards for most of the devices. Is is some way of saying "This is the default config for all DeathAdder mice" or something similar.

The new automatic config that is supposed to go somewhere in /var will not have these wildcards. It will be configurations for actual mice that the user actively configured. So this config, if it exists, would override the /etc config.

This can't be stored in the user's home directory. razerd is a global daemon run by root (it has to).
Unless you want to introduce a _third_ way of configuring: User configurable manual configuration sets.
This would be a nice feature, too, but these sets can't be auto-loaded on startup or hotplug by design. So this wouldn't solve your original issue.

rcmn, 2015-09-07:
Well it sounds good then to have razer.conf as the default config file. But right now to exist the user have to create it.It did not exist until I copy\paste from the repo. Is it something that will change in the future , where the install or the demon will create it ? 
And now I understand why you need to have the new automatic config in /var. Implementing a third way seems a bit of an overkill.

polyzen, 2015-09-07:
How are you installing razer?

mbuesch, 2015-09-07:
The distro package is supposed to install it. I think the debian package should do that already. The cmake currently doesn't do so, because that /var thingie doesn't exist, yet. I don't want to overwrite random user configs, which _currently_ might reside in /etc

rcmn, 2015-09-07:
I did use the cmake . I did not find or look to hard to find a deb package.
So that explains it .

Muson117, 2015-09-10:
Hi,

So is there a way to dump qrazercfg config to /etc/razer.conf?
Default /etc/razer.conf is missing options (deathadder chroma) that are available on qrazercfg like setting dpi to 500 (in conf it's 800 minimum). Just setting plain 500 in conf file ignores file completely.

If not, what is the purpose of qrazercfg tool?

mbuesch, 2015-09-10:
>So is there a way to dump qrazercfg config to /etc/razer.conf?

No, not yet.

>If not, what is the purpose of qrazercfg tool?

It is the graphical frontend of razercfg.
That doesn't have anything to do with config saving or restoring. Why does this come up all the time?

Muson117, 2015-09-10:
Good work Michael, :+1: 
I really appreciate your effort in razerd work.

Me and I guess 99.9% of your software (qrazercfg) users don't fancy setting up mouse config on every boot, every day, every week, sometimes even few times a day.... I think life is a bit too short to adjust dpi everytime.

Imagine Adobe Photoshop or LibreOffice package without a save button... what a useless products would they be... used only to torture people in inability to save... :) 
So I was checking maybe I don't understand TRUE purpose of qrazercfg, maybe qrazercfg purpose is entertainment... but it kind of wears of on 3rd try... 
If you used it only as python exercise (learning) and it has no actual real life application it's totally fine with me. :)

mbuesch, 2015-09-10:
I think we have a little misunderstanding here, Muson117.

I am _totally_ with you for implementing a settings restore functionality. But it does _not_ belong into qrazercfg. It belongs into razerd. We discussed that earlier here.
And in addition to that, you _don't_ need to reconfigure your settings after every boot _today_ (or even for years). Just put your initial config into /etc/razer.conf and you are done. razerd will apply that. Yes, the config parser has a few unimplemented features, but most of the time that won't hurt.

razercfg is not a toy software like Photoshop. See here for the software development basics that apply to it: http://bues.ch/cms/resources/contribute.html
But it also is Free Software. That means if you want some feature, the best way to get it is to actually implement it (obeying the rules above). I would be very happy to merge that feature.

Muson117, 2015-09-11:
Thanks for clarifying, can you answer one more question.
For instance in default razer.conf there are (DA chroma):

	# Configure first profile
	# Resolution: 800, 1800, 3500, 5600, 10000
	res=1:800

in qrazercfg I can set resolution to 400, but if I change line to res=1:400 whole mouse config is ignored after razerq restart.

Can I somehow lower resolution than 800 with razer.conf?

Many thanks

mbuesch, 2015-09-11:
The config currently only supports changing dpi-mappings, but not modifying one dpi mapping. So that currently is not implemented.

z3ntu, 2015-10-06:
So how can I exactly save my DPI somewhere that it loads it on startup? I only used qrazercfg for configuration so far.

polyzen, 2015-10-07:
/etc/razer.conf

z3ntu, 2015-10-07:
And what do I have to write into that?

z3ntu, 2015-10-13:
@polyzen Can you please explain me what to save info /etc/razer.conf ?

Muson117, 2015-10-13:
Hey, there is demo conf file in installation dir, copy to /etc, rest is self explaining and uncomment what is needed, except why qrazercfg was created at all ;-)

gozzilli, 2015-11-28:
This is interesting, and would be a welcome feature indeed. 
Is there a workaround in the meantime? Is there a config file that we could edit manually to achieve this? I can't see any `/etc/razer.conf` on my machine. 

Many thanks!

mbuesch, 2015-11-28:
>I can't see any /etc/razer.conf on my machine. 

Well, you can create it. An example is shipped with razercfg.

gozzilli, 2015-11-28:
> I can't see any /etc/razer.conf on my machine.

Sure, I can do that. Thanks.

mjtorn, 2016-01-09:
To be sure I got this straight; what's needed is a `razer_save_config(const char *configfile)` function in razerd.c, which can be accessed over the control socket (by eg. qrazercfg) to save a more permanent copy of the settings?

It would also seem that the mouse has internal RAM, no battery-backup or permanent flash, for storing these updates, right?

How does updating the mouse work, by the way? I see USB traffic when I start razerd and sometimes when I toggle the DPI or LED settings, but not always. Is this due to the kernel or someone buffering the output of USB traffic before showing it? I had the similar phenomenon before when I tried the OrbWeaver and would really like to understand this.

This is also quite visible when choosing `Re-apply all hardware settings` from the menu while looking at the USB traffic; sometimes nothing shows up. AFAICT the code path from sending `COMMAND_ID_RECONFIGMICE` winds up at `naga_usb_write` which has no checking for something like "don't bother updating, nothing changed".

mbuesch, 2016-01-09:
>To be sure I got this straight; what's needed is a razer_save_config(const char *configfile) function in razerd.c, which can be accessed over the control socket (by eg. qrazercfg) to save a more permanent copy of the settings?

First of all there needs to be a new mechanism to retrieve the settings from the actual hardware layer. Currently there is no such thing. razerd/librazer just writes the settings to the hardware layer. The other way around is not designed/implemented.

>It would also seem that the mouse has internal RAM, no battery-backup or permanent flash, for storing these updates, right?

There are devices with permanent storage and without permanent storage.
But this is not a hardware issue. The thing needs to be solved in the hardware layer of librazer.

> I see USB traffic when I start razerd and sometimes when I toggle the DPI or LED settings, but not always. 

You should always see traffic.

mjtorn, 2016-01-09:
>  But this is not a hardware issue. The thing needs to be solved in the hardware layer of librazer.

So we need to get the current settings to display them correctly, but also have a writable settings file so we can change the hardware, and then of course query it later?

Do you literally mean "First of all"? Because my approach would be to simply update the configuration file first, _only then_ add reading from the hardware. The reason is pragmatic; it gets the job done. It's nice to know what a piece of hardware is configured to, but I'd bet 20 euros that the most common use case is "I booted my computer and would like to restore the settings from /etc/razer.conf automatically".

> You should always see traffic.

I still have it on my roadmap to look into Wireshark and friends. What you say and what the code apparently says does not add up with what I'm seeing, unless there's terminal-flushing logic that's keeping me from seeing events in real-time. I should double-check this at some point to reading the USB events in some other way, but let's see when...

mbuesch, 2016-01-09:
>only then add reading from the hardware. 

The settings cannot be read from hardware (on all devices). And it's not necessary to do that. The hardware drivers already have all information that is needed. It just has to be exposed to librazer/razerd and _then_ it can be written to the file. This is the part that is missing. There is no way for retrieving the current configuration from the hardware layer.

>but I'd bet 20 euros that the most common use case is "I booted my computer and would like to restore the settings from /etc/razer.conf automatically".

There are no changes required for this.
The restore part is there. What is missing is the saving part.

>I still have it on my roadmap to look into Wireshark and friends.

I highly suggest that you do that. Wireshark pretty easy to use and at the same time really powerful.

john-2000, 2016-05-17:
Sorry but I am probably a bit slow her, I have been trying to get my deathadder chroma to change settings to move from spectrum to fixed colours (and if possible the DPI - I am little unsure about the comments following https://github.com/mbuesch/razer/issues/49#issuecomment-147638910 )

my /etc/razer.conf looks like this:

`
# Razer DeathAdder Chroma mouse
[Mouse:DeathAdder Chroma:*:*]
	# Config section disabled?
	disabled=false

	# Configure first profile
	# Resolution: 800, **1500**, **2200**, 5600, 10000
	res=1:**1500**
	# Frequency: 125, 500, 1000
	freq=1:500

	# Initial profile selection
	profile=1

	# Configure LEDs
	led=1:GlowingLogo:on
	mode=1:GlowingLogo:static
	color=1:GlowingLogo:bbbb00
	led=1:Scrollwheel:on
	mode=1:Scrollwheel:static
	color=1:Scrollwheel:bbbb00
`

could you help me get this right? I seem to need to run qrazercfg to adjust the DPI and run `razercfg -m Scrollwheel:static -m GlowingLogo:static -c GlowingLogo:bbbb00 -c ScrollWheel:bbbb00
` to adjust the colours

thanks

mbuesch, 2016-05-17:
@john-2000 
That is not related to this issue. Please open another one, if you need further help.

Apart from that your config should be fine.
If you installed razercfg correctly, it should load these settings on boot and hotplug.

tominglis, 2016-12-12:
I installed qrazercfg on Ubuntu 16.04 using the instructions here:

http://www.webupd8.org/2016/06/configure-razer-mice-in-linux-with.html

I have read the readme and most of the discussions on Github.

I am a little confused about how to persistently store settings correctly? If I run qrazercfg (as a user - not sure if I am supposed to do this as root?), the settings are not saved between reboots.

I have tried: 

"Reapply all hardware settings"

and manually editing the /etc/razer.conf file:

# Razer DeathAdder Chroma mouse
[Mouse:DeathAdder Chroma:*:*]
	# Config section disabled?
	disabled=false

	# Resolution: 800, 1800, 3500, 5600, 10000
	res=1:1800
	res=2:800
        res=3:1800
        # Frequency: 125, 500, 1000
	freq=1:1000
        freq=2:500
        freq=3:500
	# Initial profile selection
	profile=1
	# Configure LEDs
	led=1:GlowingLogo:on
	led=1:Scrollwheel:on
	#mode=1:Scrollwheel:static
	#color=1:Scrollwheel:0000FF

It seems to save the settings now, but I am not sure which method I used was correct?

seantcanavan, 2017-09-25:
@tominglis I have the same exact problem. Apparently it involves writing your settings to `/etc/razer.conf` but my mouse, Razer Mamba Tournament Edition, is not listed in the list of mice inside that configuration.

KonradHoeffner, 2020-06-08:
@seantcanavan: Can you create your own section? I can get the identifier for my mouse using `razercfg -s`, which results in `Mouse:DeathAdder 3500DPI:USB-002:1532-0016-0`.
The entry in `/etc/razer.conf` is titled with `[Mouse:DeathAdder*:*:*]`.

seantcanavan, 2020-06-08:
@KonradHoeffner I gave up on this a *long* time ago. I seem to remember downloading more and more third party utilities that were sniffing USB protocol messages trying to hack settings changes back into the mouse. After soft-bricking my mouse I realized it wasn't worth the effort.