mbuesch/razer #52: Gradient spectrum support

krochonw, 2015-12-03:
Would it be possible to add something like a gradient fade mode?

mbuesch, 2015-12-04:
Could you elaborate a bit, please? What do you mean by fade mode? Fading for what?

Thanks.

krochonw, 2015-12-04:
on the colour modes, static, spectrum and off, it would be cool if you could do something similar to spectrum but be it fading back and forth between 2 colours as a gradient. I know its not a normal feature on the mice but was wondering if it is possible. 
Thanks for the reply.

mbuesch, 2015-12-04:
Ok. Thanks for your suggestion, but that is not possible to do. Razercfg implements what the hardware supports. If the hardware doesn't support some type of LED fading, razercfg can't do it either.

closed 2015-12-04 by mbuesch