mbuesch/razer #54: How to do the intial headset support?

z3ntu, 2016-01-03:
Hey,
I want to try to add the Razer Kraken Chroma (Headset) as supported device to this software, mostly for controlling the colors.
I already began working on the support (which is kinda hard because I have little to none experience with USB drivers and C programming ^^, and the software is not made for headsets yet). Repo/fork is here: https://github.com/z3ntu/razer .
Well at least I already know what data the software should send to the headset.
Can you tell me where I have to add support too, that the headset gets detected? By now I just know that the mainloop of razerd calls "razer_rescan_mice". Is it right to add another 'or' to https://github.com/mbuesch/razer/blob/master/librazer/librazer.c#L891 ?

mbuesch, 2016-01-03:
The main API is in librazer/librazer.h. Everything is built around that. So you need to add a new device type there and add structures and operations for your added features. Note that the color structures already are device agnostic, so you can reuse them in a headset device type.
If you got that, you can extend razerd (which uses the librazer API) and the user python APIs.

>mainloop of razerd calls "razer_rescan_mice". Is it right to add another 'or' to

Well, no. Just like the name says, this librazer API function is only supposed to scan for mice.
You should probably add a new razer_rescan_headsets() to scan for your new device type.

You basically need to touch every part of the whole stack to add the new device type into the APIs and protocols. There are a lot of things that can be reused (like struct razer_led and friends), but the basic new device structures need to be designed from scratch.

z3ntu, 2017-01-01:
Closing this as I have switched to [razer-drivers](https://github.com/terrycain/razer-drivers) which now has Kraken headset support.

closed 2017-01-01 by z3ntu