mbuesch/razer #66: Consider uploading this to Debian

medicalwei, 2016-03-14:
Hello.

As you have prepared debian packaging script, I am wondering if you are willing to upload this package to Debian.

If you don't have time or unwilling to do so, I could also help as I also need this to make my DeathAdder Chroma not so sensitive.

mjtorn, 2016-03-14:
I added a bunch of the `debian/` stuff that's currently in here, and will say straight up that I do not have time for the New Maintainer process, and actually not too much interest in maintaining a Debian package. I made the patches just to get this installed nicely under SteamOS.

Once I get settled into my new apartment I hope I'll have time soon to set up a repository for SteamOS packages, and this would definitely be there, but finding a place, dealing with the paperwork, the actual move, and unpacking boxes etc will take a short eternity. I haven't even tested the internet connection, much less hooked up my desktop or Steam Machine.

mbuesch, 2016-03-14:
>if you are willing to upload this package to Debian.

I'm not a DM or DD, so I can't.

>I could also help

So you are DM or DD?
If so, feel free to make a Debian package. If you make any changes to the Debian scripts of the source code itself, please submit them upstream so I can apply them here, too.

medicalwei, 2016-03-15:
I am NOT a DM nor a DD. But packaging is one of the common steps toward it. 

A person who is not a DM or DD requires sponsored upload; it still needs a help from DD to upload the package. If you want to upload the package, and know someone whom are DD, you could ask them for upload.

mbuesch, 2016-03-15:
Well, OK. Anybody feel free to look for a sponsor. I'm currently not planning to do that.

bluca, 2017-10-25:
Hi, I've taken ownership of https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=587987 and uploaded razercfg in Debian. It's pending review in the NEW queue.

The debian/ repo is at: https://github.com/bluca/razercfg-debian
Delta is as small as possible, only for necessary bits: Maintainer/Vcs fields in d/control, quilt format rather than native, and +ds repack of the tarball to remove debian/ via watch file.

The other difference is DH compat level 10, which makes it incompatible with Debian 7 so I've not pushed it via https://github.com/mbuesch/razer/pull/112 yet, will do once Wheezy goes EOL to avoid causing problems for users rebuilding locally.

mbuesch, 2017-10-26:
@bluca 
Thanks a lot for your great work.
Please keep contributing the debian/ changes upstream, if there are any. (I would rather like to keep native format though)

blshkv, 2018-03-23:
I would suggest to remove ./debian folder and close this bug. Debian can patch it as they need it but a software should not distribute a special files for each distribution.

bluca, 2018-04-19:
It's been accepted, so it will be available in sid tomorrow and in testing in 5 days. I'll upload to stretch-backports as well next week. So I think this issue can be closed now.

mbuesch, 2018-04-19:
Thanks a lot.

closed 2018-04-19 by mbuesch