mbuesch/razer #67: DA Chroma logo LED color changing isn't working

medicalwei, 2016-03-14:
Hi,

I try to fiddle with the LEDs of the mouse. While the scroll wheel LED does change, the LED on the logo doesn't.

What diagnose steps do I need to take? I can try to use another computer with Windows/Mac though.

z3ntu, 2016-03-14:
I actually have the Deathadder Chroma and it worked for me some time ago...

mbuesch, 2016-03-14:
I'm not aware of any issues here.

@medicalwei  and @z3ntu What firmware versions do you have?

z3ntu, 2016-03-14:
Just tested it with `qrazercfg` and everything works as expected. I have the Deathadder Chroma `model no. RZ01-0121`.
Also using `razercfg -c GlowingLogo:666666` in the terminal works.

medicalwei, 2016-03-15:
Same model no. (RZ01-0121) and firmware version (3.01) here.

medicalwei, 2016-03-15:
Sorry but I am considering this problem of my mouse now. The error message does not come out, but the glowing logo does not glow. Trying on the Mac driver.

medicalwei, 2016-03-15:
Seems that I turned the LED off using the official driver, and I cannot turn it on using razercfg.
LED color changing works after turning the LED back on using synapse.

mbuesch, 2016-03-15:
Ok, the only way to debug this is to look at the USB traffic and see what we are doing differently from the original driver. Wireshark can be used for this.
I don't have the hardware, so I can't do that myself.

medicalwei, 2016-03-21:
Sorry for the late update. The captured traffics of the original driver is uploaded below:
[DAChroma-USBPcaps.zip](https://github.com/mbuesch/razer/files/181721/DAChroma-USBPcaps.zip)

Please don't hesitate if you have further need of testing/capturing/etc.

medicalwei, 2016-03-21:
And here's the USB traffic capture from razercfg. 
[razercfg-dachroma.zip](https://github.com/mbuesch/razer/files/181730/razercfg-dachroma.zip)

kzemek, 2016-03-22:
A bit of insight: when I wrote the Chroma driver, Synapse LED toggle (or indeed any other setting if I remember correctly) had no permanent effect on the mouse. It may be a capability added in a later firmware upgrade, with old opcodes preserved for compatibility but the permanent "off" state taking precedence over temporary states that razercfg sets.

john-2000, 2016-04-25:
Hi
I have a DeathAdder Chroma USB 
lsusb: _ID 1532:0043 Razer USA, Ltd_ 
Firmware version 5.02 
and I am experiencing the same thing -- ie I can set LED from the qrazercfg tool but not from CLI razercfg.
Is there anything I can do to help to get this fixed?

kzemek, 2016-04-25:
@john-2000 This seems like a different issue, unless @medicalwei can change the LED color from qrazercfg?

john-2000, 2016-04-25:
Ok, this is odd it now works from the command line - I added a razer.conf file by hand and now:
`razercfg -c GlowingLogo:ffff77
razercfg -m ScrollWheel:spectrum
razercfg -c ScrollWheel:ff0000` 

all work just fine -- thanks for the great tool !! (and thanks for the fast response)

mbuesch, 2016-07-09:
I'm closing this, because there doesn't seem to be a way to find the cause of the original issue.
If the problem re-appears, please re-open this issue.
Thanks!

closed 2016-07-09 by mbuesch