mbuesch/razer #68: Setting Environment for OS with systemd

deekej, 2016-03-16:
After installation, trying to enable the razerd (via 'systemctl start razerd') will fail on Fedora, because the service can't find the librazer.so.1.

The easiest way to fix it is to change the razer.service.template - here's the diff:
[diff.txt](https://github.com/mbuesch/razer/files/175896/diff.txt)

Alternatively, you can mention this in the READM.md file - if this happen to any user, he should add the line below into the razer.service.template and run the make again.

Environment=LD_LIBRARY_PATH=/usr/local/lib

z3ntu, 2016-03-16:
The problem is that `librazer.so.1` is not guaranteed to be in `/usr/local/lib`

mbuesch, 2016-03-16:
>Environment=LD_LIBRARY_PATH=/usr/local/lib

No. This is not a correct fix.
If you install razercfg to /usr/local it is your responsibility to make sure the installed libraries can be found.

```make install``` will already run ldconfig. If you installed it somewhere, where ldconfig cannot find it, it's a configuration error of your operating system or you installed it to the wrong place.
If your operating system ships an /etc/ld.so.conf*, which includes /usr/local/lib, everything is fine. It your operating system doesn't ship that and you don't want to add that to ld.co.conf, simply don't install razercfg to that location.

So I don't see this as an issue of razercfg.

z3ntu, 2016-03-16:
@mbuesch Maybe it would be good to add this as a note to the readme (as it gets installed to `/usr/local` without further configuration)

mbuesch, 2016-03-16:
@z3ntu I think a sane operating system should already have /usr/local/lib in its ld.so.conf.
But you are right. We could add a small note to the readme. Perhaps:
```
Be aware that make install installs the shared library librazer.so to ${install_prefix}/lib.
You you have to make sure that it can be found there by the dynamic linker ld.so.
This can be done via /etc/ld.so.conf or /etc/ld.so.conf.d/.
See your operating system manual for further information.
```

z3ntu, 2016-03-16:
Maybe also that it is installed to `/usr/local/` by default, and maybe that you can change the prefix with `-DCMAKE_INSTALL_PREFIX='<somewhere>'`.

mbuesch, 2016-03-16:
@z3ntu OK. Feel free do do a pull request.

z3ntu, 2016-03-16:
I actually have no idea how to phrase that :D

mbuesch, 2016-03-16:
```
Be aware that make install installs the shared library librazer.so to $PREFIX/lib.
The default $PREFIX is /usr/local/, but the install prefix can also be changed via -DCMAKE_INSTALL_PREFIX='<somewhere>'
You you have to make sure that librazer.so in $PREFIX/lib/ can be found by the dynamic linker ld.so.
Your operating system most likely already has support for libraries in /usr/local/lib. So on most systems you don't have to do anything.
If this is not the case, or you installed razercfg somewhere else, a new library search path can be added via /etc/ld.so.conf or /etc/ld.so.conf.d/.
See your operating system manual for further information.
```

?

mbuesch, 2016-03-16:
Closing this issue, because we added a note to the README.

If you still think this is a problem and should be fixed in another way, re-open please.
Thanks a lot for your report.

closed 2016-03-16 by mbuesch

deekej, 2016-03-19:
No, I think this is a good solution. :) Thank you very much!