mbuesch/razer #71: [GUI]Problems with latest git source

presianbg, 2016-03-28:
Hi, 

With the latest git source I have following error when I try to launch qrazercfg.

Traceback (most recent call last):
  File "/usr/local/bin/qrazercfg", line 22, in <module>
    from pyrazer import ui
ImportError: cannot import name 'ui'

Could you check this out?

Thank you in advance.

presianbg, 2016-03-28:
Looks like a path issue. I resolved that by copying "ui.py":

`sudo cp /usr/local/lib/python3.4/site-packages/pyrazer/ui.py /usr/lib/python3.4/site-packages/pyrazer/`

mbuesch, 2016-03-28:
> Looks like a path issue. 

Right. :)

closed 2016-03-28 by mbuesch

z3ntu, 2016-03-28:
@presianbg You could also install razercfg to /usr instead of /usr/local, then the paths should all be correct.

mbuesch, 2016-03-29:
To clarify this for other people having the same issue:
Make sure to uninstall any old razercfg (use uninstall.sh), before you install razercfg to a new prefix. Even if you install a new version of razercfg to the same prefix as before, it's a good idea to run uninstall.sh before doing so, to remove old files.

presianbg, 2016-03-29:
Ok. I uninstalled it all. And make clean install. But still have some weird path issues. The razerd is working as expected, but when i try to run qrazercfg:

`Traceback (most recent call last):
  File "/usr/local/bin/qrazercfg", line 21, in <module>
    import pyrazer
ImportError: No module named 'pyrazer'
`

And this are the files with their absolute paths:

`/usr/lib/systemd/system/razerd.service
/usr/lib/udev/rules.d/80-razer.rules
/usr/local/bin/qrazercfg
/usr/local/bin/razer-gamewrapper
/usr/local/bin/razercfg
/usr/local/bin/razerd
/usr/local/lib/librazer.so
/usr/local/lib/librazer.so.1
/usr/local/lib/python3.4/site-packages/pyrazer
/usr/local/lib/python3.4/site-packages/razercfg-0.33-py3.4.egg-info
/usr/local/lib/python3.4/site-packages/pyrazer/__init__.py
/usr/local/lib/python3.4/site-packages/pyrazer/__pycache__
/usr/local/lib/python3.4/site-packages/pyrazer/main.py
/usr/local/lib/python3.4/site-packages/pyrazer/ui.py
/usr/local/lib/python3.4/site-packages/pyrazer/__pycache__/__init__.cpython-34.pyc
/usr/local/lib/python3.4/site-packages/pyrazer/__pycache__/main.cpython-34.pyc
/usr/local/lib/python3.4/site-packages/pyrazer/__pycache__/ui.cpython-34.pyc
/usr/local/share/applications/razercfg.desktop
`

I'm on Fedora 23 (if that helps).

mbuesch, 2016-03-29:
Is your system's Python configured to look for modules in /usr/local?

presianbg, 2016-03-29:
I'm not sure. I think it's looking in /usr/lib/python3.4 and /usr/lib64/python3.4 . When i do:
`
sudo cp -R /usr/local/lib/python3.4/site-packages/pyrazer /usr/lib/python3.4/site-packages/
`
Everything works as expected.

mbuesch, 2016-03-29:
You can use the PYTHONPATH environment variable to add paths to the module search path.

z3ntu, 2016-03-29:
If you want to use another prefix, use `/usr`, as everything will go into the correct directories (don't use `/usr/lib`!)