mbuesch/razer #72: Main website is offline

9Morello, 2016-03-29:
http://bues.ch/cms/hacking/razercfg.html is offline.

z3ntu, 2016-03-29:
If you want to access it right now, you can use the google cache: http://webcache.googleusercontent.com/search?q=cache:http://bues.ch/cms/hacking/razercfg.html&gws_rd=cr&ei=Xob6Vo71F8izabX7j7gK

mbuesch, 2016-03-29:
Yes, I'm sorry. I updated some software and it broke my CMS config.
I'm working on it. It will be available soon.

mbuesch, 2016-03-29:
Everything related to razercfg should be working correctly again.
It might be possible that there will be small interruptions ranging up to a few minutes each in the next few days though.
Thanks for your report. If there still is something that doesn't work correctly, I'd be happy to get your report again.

closed 2016-03-29 by mbuesch