mbuesch/razer #73: Set custom resolution on Deathadder Chroma

Algram, 2016-04-04:
Hello, 
I wanted to set a custom resolution to my Deathadder Chroma with `razercfg -r 1:2400x2400`, but I get an "Invalid resolution" error when trying to do so.

How do I set a resolution on my mouse that is not one of the default resolutions?

Thank you for your help!

mbuesch, 2016-04-04:
This is a known limitation of the current config handler.
See https://github.com/mbuesch/razer/issues/49#issuecomment-139571978

mbuesch, 2016-06-30:
A better description of the missing feature:
https://github.com/mbuesch/razer/issues/76#issuecomment-229713747

DanielPower, 2017-07-23:
So is there no way to set dpi to a custom value without using qrazercfg? It seems odd that the gui contains features that the cli tool cannot use.

I use 1100 as my dpi for playing Countrer-Strike, but razercfg only lets me choose either 800 or 1500 which is either too low or too high. In order to set the mouse dpi properly, I have to manually change it in qrazercfg every time I start my system, since I can't set a custom dpi in /etc/razer.conf either.

mbuesch, 2017-07-23:
> So is there no way to set dpi to a custom value without using qrazercfg? It seems odd that the gui contains features that the cli tool cannot use.

Modifying a DPI mapping is not implemented in the cli tool and the configuration parser, yet.
Please feel free to implement these new features and send in a pull request for discussion.

But beyond that your mouse should keep the setting, if it contains persistent config storage. Which all modern mice do.

DanielPower, 2017-07-23:
> But beyond that your mouse should keep the setting, if it contains persistent config storage. Which all modern mice do.

I have a Deathadder Chroma, not sure which year, but it's the 10,000 DPI model. It does not store the settings. Every time I reboot my computer, or restart razerd, all of the profiles have reverted to default. So I have to manually startup qrazercfg and edit my profiles every time I start my computer.

I'm taking a look through the razercfg code right now, will try to add the functionality in the next day or two, since I have some time off.

DanielPower, 2017-07-23:
I've written support for a `--mapping` option, which will modify the resolution for the currently used mapping on the currently used profile. This solves my issue with not being able to set a custom resolution with razercfg, but does not solve the issue of setting a custom resolution via /etc/razer.conf.

janstuemmel, 2017-07-24:
>> But beyond that your mouse should keep the setting, if it contains persistent config storage. Which all modern mice do.

> I have a Deathadder Chroma, not sure which year, but it's the 10,000 DPI model. It does not store the settings. Every time I reboot my computer, or restart razerd, all of the profiles have reverted [...]

mine too. its more about when the razer deamon starts then it reapplies the defaut settings

mbuesch, 2017-07-24:
>mine too. its more about when the razer deamon starts then it reapplies the defaut settings

Yes, this might be a shortcoming of the choma driver, or the device itself. I don't know.
It depends on whether there is a command to read the device's configuration.

As a simple workaround: Don't start razerd automatically.