mbuesch/razer #74: lachesis set led mode and color issue

LawrenceBobrov, 2016-05-05:
Hi,guys! Thank you very much for this cool driver pack!
Actually, everything works almost fine with my lachesis (firmware v. 1.94) in Ubuntu 14.04 x64.
I can switch on and off both ScrollWheel and GrowlingLogo. But I can not change led mode and color.
I tryed to do it (1)on the fly, (2) turning off LED, setting up color, and turning led LED on, (3)restarting services,etc.
It seems like a bug, but I do not have enough knowledge to debug this driver issue myself. Could you,please,help?

mbuesch, 2016-05-10:
>But I can not change led mode and color.

Is that a Lachesis 5600 DPI?
If so, what does "cannot change" mean?

LawrenceBobrov, 2016-05-11:
Hello,@mbuesch! Thank you for your reply!

> Is that a Lachesis 5600 DPI?

No. It is not. 

> what does "cannot change" mean?

I try to modify color and led modes with the following commands: 
`razercfg -c Scrollwheel:FF0000`
`razercfg -c GlowingLogo:ffff00`
`razercfg -m Scrollwheel:breathing`
`razercfg -m GlowingLogo:spectrum`
But they do not affect mouse at all. As I mentioned above I also tried to turn led off before changing color, restarting razerd service, etc.

But the commands for turning leds on and off work fine:
`razercfg -l Scrollwheel:off`
`razercfg -l Scrollwheel:on`
`razercfg -l GlowingLogo:off`
`razercfg -l GlowingLogo:on`

z3ntu, 2016-05-11:
@LawrenceBobrov Which "Lachesis" is it (as it is not the 5600DPI variant)?

LawrenceBobrov, 2016-05-11:
@z3ntu 
This is Lachesis with 4000dpi which is called "Lachesis Classic" in razercfg.
http://www.razersupport.com/gaming-mice/razer-lachesis/

mbuesch, 2016-05-11:
>This is Lachesis with 4000dpi which is called "Lachesis Classic" in razercfg.

Are you sure this one has hardware support for color leds?
My Lachesis classic does not have that.

Can you please tell us the USB IDs and the firmware version of your device?

LawrenceBobrov, 2016-05-12:
@mbuesch 
Actually, I'm not sure about it. I recently got this mouse from my friend (he did not use it for a long time) but he said it could probably change color.
```
razercfg -V
Mouse:Lachesis Classic:USB-001-002:1532-000C-0: Firmware version 1.94
```

It looks like lsusb information is included, but I'm not 100% sure. So:
```
lsusb |grep Razer
Bus 001 Device 002: ID 1532:000c Razer USA, Ltd
```

mbuesch, 2016-05-12:
It's pretty certain that this device does not have color LEDs.
So everything is fine.
I'm closing the issue. If there still is some problem, please re-open.

closed 2016-05-12 by mbuesch

LawrenceBobrov, 2016-05-12:
@mbuesch 
Thank you very much and sorry for this situation with colors.
However, there is still an issue with LED modes as I mentioned above.
Should this mouse support them?

mbuesch, 2016-05-12:
>Should this mouse support them?

Sorry for the confusion.
The Lachesis classic doesn't support LED mode changes either.

LawrenceBobrov, 2016-05-13:
@mbuesch 
Thank you for the clarification!