mbuesch/razer #75: DeathAdder 1800 not supported

fernandocamargoti, 2016-05-11:
When I was executing "razercfg -s", not was showing on my terminal. After reading other issues, I noticed that all supported mouses were insite librazer.c. So, since I think my mouse is completly compatible with the 2013 version, I added the following line that to solve my problem:

USB_MOUSE(0x1532, 0x0038, &razer_deathadder2013_base_ops), /* 1800 edition */

After that, razercfg recognizes my mouse as if it's the 2013 edition and everything works well. I didn't submit a pull request adding this line becouse I think it's not the optimal solution. But if you want, I can do it.

mbuesch, 2016-05-15:
Please capture the bus communication with the original Windows driver and look for differences in the protocol.

Enigma0, 2016-08-30:
Same problem here. It works with the 2013 DeathAdder but the settings aren't remembered. Also I am seeing this:

```
sudo service razerd status
● razerd.service - Razer device state daemon
   Loaded: loaded (/lib/systemd/system/razerd.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2016-08-29 21:26:55 EDT; 35min ago
  Process: 2467 ExecStart=/usr/bin/razerd -f -B -P /run/razerd/razerd.pid (code=exited, status=0/SUCCESS)
 Main PID: 2481 (razerd)
   CGroup: /system.slice/razerd.service
           └─2481 /usr/bin/razerd -f -B -P /run/razerd/razerd.pid

Aug 29 21:26:55 SteamMachine systemd[1]: Starting Razer device state daemon...
Aug 29 21:26:55 SteamMachine razerd[2481]: Razer device service daemon
Aug 29 21:26:55 SteamMachine systemd[1]: Started Razer device state daemon.
Aug 29 21:26:58 SteamMachine razerd[2481]: librazer: razer-deathadder2013: Failed to read firmware version
```

rainbowjose, 2016-10-14:
I've done this trick too, but my mice didnt appeared. Without logs or something ots very hard to research whats wrong.

fernandocamargoti, 2017-04-10:
Soon I'll try to follow your guide and configure a virtual machine with Windows to capture the the bus communication of my mouse.

rainbowjose, 2017-04-10:
https://github.com/rainbowjose/razer

fernandocamargoti, 2017-04-10:
Thank you, @rainbowjose. What have you done in your fork? The same thing I did?

blshkv, 2018-03-23:
@rainbowjose you should create a pull request and contribute your work back to the upstream so that it would be one main source only

rainbowjose, 2018-03-24:
Я не разбираюсь в этой хуйне блять, все это сложно, там две строчки кода
изменены, я в рот ебал их куда-то пушить, реквестить, я в этом ничего не
понимаю, я просто для себя форкнул когда-то.

mbuesch, 2018-03-24:
@rainbowjose, such profanity is not acceptable here, so I deleted your post.
You are welcome to contribute to this project; in English and without using swearwords.
Thanks.