mbuesch/razer #76: /etc/razer.conf apparently not loaded / applied to DA:Chroma

john-2000, 2016-05-17:
@mbuesch hi thanks for answering my other post about this - here as a new issue

Sorry but I am probably a bit slow her, I have been trying to get my deathadder chroma to change settings to move from spectrum to fixed colours (and if possible the DPI - I am little unsure about the comments following #49 (comment) )

I have re-downloaded the razer-master.zip and run the install with

> cmake .
> make
> sudo make install

installation looked sucessful but my /etc/razer.conf does not seem affect the mouse settings - ideally I would like static light in faint yellow (bbbb00) on both scrollwheel and body and a 2 resolution setting of 1500 dpi applied.

my /etc/razer.conf looks like this:

>     [Mouse:DeathAdder Chroma::]
>     # Config section disabled?
>     disabled=false
>     # Configure first profile
>     # Resolution: 800, 1500, 2200, 5600, 10000
>     res=1:1500
>     # Frequency: 125, 500, 1000
>     freq=1:500
>     # Initial profile selection
>     profile=1
>     # Configure LEDs
>     led=1:GlowingLogo:on
>     mode=1:GlowingLogo:static
>     color=1:GlowingLogo:bbbb00
>     led=1:Scrollwheel:on
>     mode=1:Scrollwheel:static
>     color=1:Scrollwheel:bbbb00

could you help me get this right? I seem to need to run qrazercfg to adjust the DPI and run razercfg -m Scrollwheel:static -m GlowingLogo:static -c GlowingLogo:bbbb00 -c ScrollWheel:bbbb00 to adjust the colours

all help is appreciated, thanks

mbuesch, 2016-06-01:
It's a bit hard for me to find out what's going on here.
Does the config work in general for you? Are other settings like DPI or led on/off applied correctly?

jediafr, 2016-06-03:
Hi,
i have a DA:Chroma on my Arch linux rig.

I'm experiencing the same issue described above : 
the /etc/razor.conf file i copied is ignored by razord.
The settings work throught the QT client though (DPI, Leds and colors).

mbuesch, 2016-06-03:
@jediafr 
> /etc/razor.conf 

A file named like this certainly needs a good shave and it's 100% correct that razerd ignores bearded files.

jediafr, 2016-06-03:
Ahah I quote your readme file : 
The user may create a razerd configuration file in /etc/razer.conf which can be used to specify various razerd options

haoanw, 2016-06-30:
@mbuesch @john-2000 for me, every setting from qrazercfg is working, but the file /etc/razer.conf is ignored. how exactly should i change it here? my file looks like this:

`...........
# Razer DeathAdder Chroma mouse
[Mouse:DeathAdder Chroma:USB-003:1532-0043-SI1608121710420]
	# Config section disabled?
	disabled=false

	# Configure first profile
	# Resolution: 800, 1800, 3500, 5600, 10000
	res=1:1000
	# Frequency: 125, 500, 1000
	freq=1:1000

	# Initial profile selection
	profile=1

	# Configure LEDs
	led=1:GlowingLogo:on
	mode=1:GlowingLogo:static
	color=1:GlowingLogo:bbbb00
	led=1:Scrollwheel:on
	mode=1:Scrollwheel:static
	color=1:Scrollwheel:0000FF

# Razer DeathAdder mouse
...........`

mbuesch, 2016-06-30:
```
 [Mouse:DeathAdder Chroma:USB-003:1532-0043-SI1608121710420]
```

Most likely the identification is wrong.
It doesn't really make sense to hard code the USB number here, because that changes all the time.
Please use globs (see the other examples).

jediafr, 2016-06-30:
Here's my section (of razer.conf) and the conf file isnt loaded....sadly
[Mouse:DeathAdder Chroma:*:*]
        # Config section disabled?
        disabled=false

        # Configure first profile
        # Resolution: 800, 1800, 3500, 5600, 10000
        res=1:1000
        # Frequency: 125, 500, 1000
        freq=1:500

        # Initial profile selection
        profile=1

        # Configure LEDs
        led=1:GlowingLogo:off
        led=1:Scrollwheel:off
        mode=1:Scrollwheel:static
        color=1:Scrollwheel:0000FF

haoanw, 2016-06-30:
now i change the device ID to `[Mouse:DeathAdder Chroma:*:*]` , it won't be loaded;
then i change the device ID to `[Mouse:DeathAdder Chroma::]`, it also won't get loaded.
but qrazercfg works fine, anyone know what is going on here?

jediafr, 2016-06-30:
no one knows,  mbuesch there is an issue there, i can get you more info if only you give me the instructions.

mbuesch, 2016-06-30:
Well, I don't have that device.

Could you please run razerd with full debugging?
```
$ sudo systemctl stop razerd
$ sudo ./razerd/razerd -l3
```

Please post the output.

jediafr, 2016-06-30:
Here you go :

sudo razerd -l3
Razer device service daemon
[razerd debug]: librazer: profile emulation: Committed active profile
[razerd debug]: librazer: Mouse profile emulation initialized for
Mouse:DeathAdder Chroma:USB-006:1532-0043-PM1501121713168
[razerd debug]: librazer: Applying config section "Mouse:DeathAdder
Chroma:*:*" to "Mouse:DeathAdder
Chroma:USB-006:1532-0043-PM1501121713168"
librazer: Config section "Mouse:DeathAdder Chroma:*:*" item "res" invalid.
librazer: Failed to apply initial config to "Mouse:DeathAdder
Chroma:USB-006:1532-0043-PM1501121713168"
[razerd debug]: librazer: Allocated and initialized new mouse
"Mouse:DeathAdder Chroma:USB-006:1532-0043-PM1501121713168"
[razerd debug]: Broadcasting mouse-add event

haoanw, 2016-06-30:
Hi, here is mine:
`viki@barney:/usr/local/bin$ sudo ./razerd -l3
Razer device service daemon
[razerd debug]: librazer: profile emulation: Committed active profile
[razerd debug]: librazer: Mouse profile emulation initialized for Mouse:DeathAdder Chroma:USB-003:1532-0043-SI1608121710420
[razerd debug]: librazer: Applying config section "Mouse:DeathAdder Chroma:*:*" to "Mouse:DeathAdder Chroma:USB-003:1532-0043-SI1608121710420"
librazer: Config section "Mouse:DeathAdder Chroma:*:*" item "res" invalid.
librazer: Failed to apply initial config to "Mouse:DeathAdder Chroma:USB-003:1532-0043-SI1608121710420"
[razerd debug]: librazer: Allocated and initialized new mouse "Mouse:DeathAdder Chroma:USB-003:1532-0043-SI1608121710420"
`

mbuesch, 2016-06-30:
```
librazer: Config section "Mouse:DeathAdder Chroma::" item "res" invalid.
```

This is not a bug, but a known missing feature.
See https://github.com/mbuesch/razer/issues/73 and https://github.com/mbuesch/razer/issues/49#issuecomment-139571978

Please continue the discussion in https://github.com/mbuesch/razer/issues/73, if you want to implement this feature.
So I'm going to close this issue, because we already have (at least) one issue for this topic.

closed 2016-06-30 by mbuesch

jediafr, 2016-06-30:
Thanks Michael, so what is the remedy ? 
comment off the res ?

haoanw, 2016-06-30:
thanks for your reply, i set res to 800 and it worked, still don't understand it fully:(

mbuesch, 2016-06-30:
>comment off the res ?

Either that, or specify a res that already has an existing mapping. You can change the mappings to your desire in qrazercfg and load one of these existing mappings via razer.conf or razercfg.

mbuesch, 2016-06-30:
>still don't understand it fully:(

DPI dual thingie on some Razer mice.
There always is a table of DPI values to select from. But for some mice (like the DA Chroma) there's also the possibility to modify the table entries. So you can basically create a table that you can later select from. That currently is only supported by qrazercfg.
If the res you specified is not in that table already, razer.conf and razercfg will fail.

haoanw, 2016-07-01:
> DPI is a dual thingie on some Razer mice.
There always is a table of DPI values to select from. But for some mice (like the DA Chroma) there's also the possibility to modify the table entries. So you can basically create a table that you can later select from. 
thanks for your reply, now i seem understand better. it seems that there is already a table of DPI mapping in razerd (5 already exist value [800] [1800] ...etc), could you please tell me where the table is stored and then i could try to modified the value inside :)

mbuesch, 2016-07-01:
>could you please tell me where the table is stored and then i could try to modified the value inside :)

The table is what you can see in the lower half of qrazercfg, below the profiles tabs and above the LEDs.

jediafr, 2016-07-02:
Thanks again for helping us...
those "possible scan resolutions" is the table.

I have to set those manually in qrazercfg each time i reboot the computer.