mbuesch/razer #77: Load custom res at boot

xoger, 2016-06-01:
Right now I'm limited to 4 default resolutions on my Razer Deathadder Chroma and trying to set anything outside of that in razer.conf or a startup razercfg command causes no configs to be loaded. Custom resolutions can be set with qrazercfg so I guess it should be possible to do this via config or via a command. So is there anyway at all to have a custom resolution set at boot?

z3ntu, 2016-06-01:
See #73 & #49

xoger, 2016-06-01:
Doesn't answer my question

z3ntu, 2016-06-01:
tl;dr: It doesn't work / isn't implemented right now.

mbuesch, 2016-06-01:
@xoger 

Thanks for your comment.

Closing this issue, because it is a duplicate of #73 and/or #49.
Please add your questions to these issues, if they are not already asked there.

closed 2016-06-01 by mbuesch

xoger, 2016-06-01:
So you wont answer why setting custom resolutions is possible with qrazercfg but not razercfg or razer.conf. ok..

z3ntu, 2016-06-01:
@xoger as I said it is not implemented yet, but if you have the C skills, you can do it and send a PR.

mbuesch, 2016-06-01:
@xoger 
>So you wont answer why setting custom resolutions is possible with qrazercfg but not razercfg or razer.conf?

It's been answered multiple times.
The shortest and easiest to understand answer was from @z3ntu:
> isn't implemented right now.

@xoger 
Feel free to implement this feature and send a pull request. This is Free Software, so you can adjust it to your needs.
And as I said, if there are any unanswered questions, please post them in one of the two already existing issues.

Thanks.

z3ntu, 2016-06-01:
@xoger This software doesn't fulfill this specific need of you right now. It is a known & wanted feature but as (as far as I know) nobody is working on it right now, it isn't a feature of this piece of software.

xoger, 2016-06-01:
I appreciate that this isn't a feature right now and I will give it a look if I have time, no promises. It just seems strange to me that a feature of the gui tool isn't available in the cl tool. My only only question is why not, if I have a clearer understanding of why this feature isn't available when you would normally expect it to be then I'll have better luck when diving into a world of unfamiliar source code. I don't mean to sound rude or demanding, I just want a clearer image of why this issue is present.

mbuesch, 2016-06-01:
@xoger 
> It just seems strange to me that a feature of the gui tool isn't available in the cl tool. My only only question is why not, 

The only possible answer is:
Because nobody implemented it. It _really_ is that easy.

Neither the config interpreter nor the CLI has support for modifying DPI mappings.
The config file format doesn't even have syntax for this, yet. So we'd first develop an actual _idea_ on how to do this in the config file and on the command line.

>expect it to be then I'll have better luck when diving into a world of unfamiliar source code.

Well, this piece of software deals with hardware. That means it's pretty complex, because it must cover a lot of different types of hardware. So the only way for you to find out how this works is to look into the source code.
This is how software is developed. You'll have to dig through the code.

xoger, 2016-06-01:
So would it be fair to say that the gui and cli were developed entirely separately?

mbuesch, 2016-06-01:
@xoger 
>So would it be fair to say that the gui and cli were developed entirely separately?

What? No.

They both have a decent software design so they don't affect each other. So there is no requirement to implement features in both symmetrically.

xoger, 2016-06-01:
so.. yes? they aren't connected? normally you'd expect the gui to run on top of the cli so you can understand my confusion. If I get time I'll give it a look but this isn't really my forte

mbuesch, 2016-06-01:
@xoger 

A quick look into the https://github.com/mbuesch/razer/blob/master/README.md would have prevented that confusion.

>normally you'd expect the gui to run on top of the cli

Almost no software does it like this.
If a project does that, it usually is due to the cli and the gui being two completely different projects. That obviously is not the case here.

xoger, 2016-06-01:
I'd like to leave this matter be but i'm confused by what you just said. Why would the gui run on top of the cli if they are separate projects? And also nothing in your readme suggests that your cli and gui are separated

mbuesch, 2016-06-01:
>And also nothing in your readme suggests that your cli and gui are separated

Oh come on, please.
See the "Architecture" paragraph and follow the ascii-art arrows. There's no arrow between razercfg and qrazercfg, so they are not based on top of each other.

xoger, 2016-06-01:
Ok thank you for clearing up my confusion