mbuesch/razer #79: not start in GUI mode

magvay, 2016-07-08:
qrazercfg

`Traceback (most recent call last):
  File "/usr/lib/python3.5/dist-packages/pyrazer/main.py", line 392, in __receive
    payload = decode(payload)
  File "/usr/lib/python3.5/dist-packages/pyrazer/main.py", line 380, in <lambda>
    decode = lambda pl: pl.decode("ASCII")
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 42: ordinal not in range(128)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/qrazercfg", line 704, in <module>
    exit(main())
  File "/usr/bin/qrazercfg", line 698, in main
    mainwnd = MainWindow()
  File "/usr/bin/qrazercfg", line 651, in __init__
    self.scan()
  File "/usr/bin/qrazercfg", line 666, in scan
    mice = razer.getMice()
  File "/usr/lib/python3.5/dist-packages/pyrazer/main.py", line 459, in getMice
    mice.append(self.__recvString())
  File "/usr/lib/python3.5/dist-packages/pyrazer/main.py", line 429, in __recvString
    return self.__receiveExpectedMessage(self.sock, self.REPLY_ID_STR)
  File "/usr/lib/python3.5/dist-packages/pyrazer/main.py", line 409, in __receiveExpectedMessage
    id, payload = self.__receive(sock)
  File "/usr/lib/python3.5/dist-packages/pyrazer/main.py", line 394, in __receive
    raise RazerEx("Unicode decode error in received payload")
pyrazer.main.RazerEx: Unicode decode error in received payload`

z3ntu, 2016-07-08:
What language/locale are you using?

mbuesch, 2016-07-08:
Which version of razercfg are you using?
Is your razercfg properly installed and does the server (razerd) version match the client (qrazercfg/pyrazer) version?

mbuesch, 2016-07-08:
And which exact type of mouse are you using? (Please tell us the USB IDs, too)

magvay, 2016-07-08:
En/en
have secondary input language Ru - but use him only for skype

magvay, 2016-07-08:
1. try use in console 
razercfg -s
Unicode decode error in received payload

2. lsusb
Bus 002 Device 004: ID 8086:0189 Intel Corp. 
Bus 002 Device 003: ID 1532:0043 Razer USA, Ltd 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 1058:0832 Western Digital Technologies, Inc. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0408:2fb1 Quanta Computer, Inc. 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
----
i try install razercfg - because i have problem with mouse
i have lag before cursor start move
if i move mouse slowly - cursor not work? if i make quick move - he start work
buttons work properly

mbuesch, 2016-07-08:
My first guess is that the serial number of your device is not all-ascii.

magvay, 2016-07-08:
hmmm 
possible change him?

z3ntu, 2016-07-08:
@magvay What input driver are you using (libinput, evdev...)?

mbuesch, 2016-07-08:
Can you please uninstall your package and try if this fixes it?
https://bues.ch/cgit/razer.git/snapshot/razer-3dcac0d1d4c0a4f462541967fa67c0e54f877349.tar.bz2

mbuesch, 2016-07-08:
The previous snapshot contains some additional issues.
Please try this instead:
https://bues.ch/cgit/razer.git/snapshot/razer-2ba796e6fa1406f57fcd0f9f065c37946879b26a.tar.bz2

magvay, 2016-07-09:
**magvay@Dell-L502X:~/Downloads/razercfg$ ls**
CMakeCache.txt       firmware              razercfg.desktop           scripts
CMakeFiles           HACKING.md            razercfg.desktop.template  udev.rules
cmake_install.cmake  icons                 razer.conf                 udev.rules.template
CMakeLists.txt       install_manifest.txt  razerd                     ui
COPYING              librazer              razerd.initscript          uninstall.sh
debian               Makefile              razerd.service
description-pak      pm-hook.sh            razerd.service.template
doc-pak              pm-hook.sh.template   README.md

**magvay@Dell-L502X:~/Downloads/razercfg$ make**
[ 83%] Built target razer
[ 95%] Built target razerd
[100%] Built target pyrazer

**magvay@Dell-L502X:~/Downloads/razercfg$ sudo checkinstall -D**
checkinstall 1.6.2, Copyright 2009 Felipe Eduardo Sanchez Diaz Duran
           This software is released under the GNU GPL.
*****************************************
**** Debian package creation selected ***
*****************************************
This package will be built according to these values: 
0 -  Maintainer: [ root@Dell-L502X ]
1 -  Summary: [ razercfg ]
2 -  Name:    [ razercfg ]
3 -  Version: [ 20160709 ]
4 -  Release: [ 1 ]
5 -  License: [ GPL ]
6 -  Group:   [ checkinstall ]
7 -  Architecture: [ amd64 ]
8 -  Source location: [ razercfg ]
9 -  Alternate source location: [  ]
10 - Requires: [  ]
11 - Provides: [ razercfg ]
12 - Conflicts: [  ]
13 - Replaces: [  ]

Enter a number to change any of them or press ENTER to continue: 
Installing with make install...
========================= Installation results ===========================
[ 83%] Built target razer
[ 95%] Built target razerd
[100%] Built target pyrazer
Install the project...
-- Install configuration: ""
-- Installing: /lib/udev/rules.d/80-razer.rules
-- udevadm control --reload-rules: 0
-- Installing: /etc/pm/sleep.d/50-razer
-- Installing: /lib/systemd/system/razerd.service
-- systemctl --system daemon-reload: 0
-- systemctl --system --force enable razerd.service: 0
-- Installing: /usr/local/share/applications/razercfg.desktop
-- Installing: /usr/local/lib/librazer.so.1
-- Up-to-date: /usr/local/lib/librazer.so
-- ldconfig: 0
-- Installing: /usr/local/bin/razerd
running install
running build
running build_py
running build_scripts
running install_lib
running install_scripts
copying build/scripts-3.5/razercfg -> /usr/local/bin
copying build/scripts-3.5/qrazercfg-applet -> /usr/local/bin
copying build/scripts-3.5/qrazercfg -> /usr/local/bin
changing mode of /usr/local/bin/razercfg to 755
changing mode of /usr/local/bin/qrazercfg-applet to 755
changing mode of /usr/local/bin/qrazercfg to 755
running install_egg_info
Writing /usr/local/lib/python3.5/dist-packages/razercfg-0.34-py3.5.egg-info
-- Installing: /usr/local/bin/razer-gamewrapper
======================== Installation successful ==========================
Copying documentation directory...
./
./README.md
./COPYING
Some of the files created by the installation are inside the home directory: /home
You probably don't want them to be included in the package.
Do you want me to list them?  [n]: 
Should I exclude them from the package? (Saying yes is a good idea)  [n]: 
Some of the files created by the installation are inside the build
directory: /home/magvay/Downloads/razercfg
You probably don't want them to be included in the package,
especially if they are inside your home directory.
Do you want me to list them?  [n]: 
Should I exclude them from the package? (Saying yes is a good idea)  [y]: 
Copying files to the temporary directory...OK
Stripping ELF binaries and libraries...OK
Compressing man pages...OK
Building file list...OK
Building Debian package...OK
Installing Debian package...OK
Erasing temporary files...OK
Writing backup package...OK
OK
Deleting temp dir...OK
**********************************************************************
 Done. The new package has been installed and saved to
 /home/magvay/Downloads/razercfg/razercfg_20160709-1_amd64.deb
 You can remove it from your system anytime using: 
      dpkg -r razercfg
**********************************************************************

**magvay@Dell-L502X:~/Downloads/razercfg$ razercfg**
Razer device configuration tool
Version 0.34
**magvay@Dell-L502X:~/Downloads/razercfg$ razercfg -s**
Failed to connect to razerd socket: [Errno 2] No such file or directory

magvay, 2016-07-09:
**magvay@Dell-L502X:~/Downloads/razercfg$ ls**
CMakeCache.txt       firmware              razercfg.desktop           scripts
CMakeFiles           HACKING.md            razercfg.desktop.template  udev.rules
cmake_install.cmake  icons                 razer.conf                 udev.rules.template
CMakeLists.txt       install_manifest.txt  razerd                     ui
COPYING              librazer              razerd.initscript          uninstall.sh
debian               Makefile              razerd.service
description-pak      pm-hook.sh            razerd.service.template
doc-pak              pm-hook.sh.template   README.md

**magvay@Dell-L502X:~/Downloads/razercfg$ make**
[ 83%] Built target razer
[ 95%] Built target razerd
[100%] Built target pyrazer

**magvay@Dell-L502X:~/Downloads/razercfg$ sudo checkinstall -D**
checkinstall 1.6.2, Copyright 2009 Felipe Eduardo Sanchez Diaz Duran
           This software is released under the GNU GPL.
*****************************************
**** Debian package creation selected ***
*****************************************
This package will be built according to these values: 
0 -  Maintainer: [ root@Dell-L502X ]
1 -  Summary: [ razercfg ]
2 -  Name:    [ razercfg ]
3 -  Version: [ 20160709 ]
4 -  Release: [ 1 ]
5 -  License: [ GPL ]
6 -  Group:   [ checkinstall ]
7 -  Architecture: [ amd64 ]
8 -  Source location: [ razercfg ]
9 -  Alternate source location: [  ]
10 - Requires: [  ]
11 - Provides: [ razercfg ]
12 - Conflicts: [  ]
13 - Replaces: [  ]

Enter a number to change any of them or press ENTER to continue: 
Installing with make install...
========================= Installation results ===========================
[ 83%] Built target razer
[ 95%] Built target razerd
[100%] Built target pyrazer
Install the project...
-- Install configuration: ""
-- Installing: /lib/udev/rules.d/80-razer.rules
-- udevadm control --reload-rules: 0
-- Installing: /etc/pm/sleep.d/50-razer
-- Installing: /lib/systemd/system/razerd.service
-- systemctl --system daemon-reload: 0
-- systemctl --system --force enable razerd.service: 0
-- Installing: /usr/local/share/applications/razercfg.desktop
-- Installing: /usr/local/lib/librazer.so.1
-- Up-to-date: /usr/local/lib/librazer.so
-- ldconfig: 0
-- Installing: /usr/local/bin/razerd
running install
running build
running build_py
running build_scripts
running install_lib
running install_scripts
copying build/scripts-3.5/razercfg -> /usr/local/bin
copying build/scripts-3.5/qrazercfg-applet -> /usr/local/bin
copying build/scripts-3.5/qrazercfg -> /usr/local/bin
changing mode of /usr/local/bin/razercfg to 755
changing mode of /usr/local/bin/qrazercfg-applet to 755
changing mode of /usr/local/bin/qrazercfg to 755
running install_egg_info
Writing /usr/local/lib/python3.5/dist-packages/razercfg-0.34-py3.5.egg-info
-- Installing: /usr/local/bin/razer-gamewrapper
======================== Installation successful ==========================
Copying documentation directory...
./
./README.md
./COPYING
Some of the files created by the installation are inside the home directory: /home
You probably don't want them to be included in the package.
Do you want me to list them?  [n]: 
Should I exclude them from the package? (Saying yes is a good idea)  [n]: 
Some of the files created by the installation are inside the build
directory: /home/magvay/Downloads/razercfg
You probably don't want them to be included in the package,
especially if they are inside your home directory.
Do you want me to list them?  [n]: 
Should I exclude them from the package? (Saying yes is a good idea)  [y]: 
Copying files to the temporary directory...OK
Stripping ELF binaries and libraries...OK
Compressing man pages...OK
Building file list...OK
Building Debian package...OK
Installing Debian package...OK
Erasing temporary files...OK
Writing backup package...OK
OK
Deleting temp dir...OK
**********************************************************************
 Done. The new package has been installed and saved to
 /home/magvay/Downloads/razercfg/razercfg_20160709-1_amd64.deb
 You can remove it from your system anytime using: 
      dpkg -r razercfg
**********************************************************************

**magvay@Dell-L502X:~/Downloads/razercfg$ razercfg**
Razer device configuration tool
Version 0.34
**magvay@Dell-L502X:~/Downloads/razercfg$ razercfg -s**
Failed to connect to razerd socket: [Errno 2] No such file or directory

AFTER REBOOT
magvay@Dell-L502X:~$ razercfg -s
Mouse:DeathAdder Chroma:USB-002:1532-0043-??????????????????????Q

magvay, 2016-07-09:
@z3ntu 
[image]
I try use google for understand what i use **libinput or evdev** and cant
Sorry I am not an experienced Linux user
How do I know my input method?

mbuesch, 2016-07-09:
@magvay 
> Mouse:DeathAdder Chroma:USB-002:1532-0043-??????????????????????Q

Interesting. So your serial number is not in ascii indeed.
What does lsusb -vv show for this device?

> 1. can i use this on github ???

I don't understand this.

> 2. can i update firmware and may be flash new sn?

You need the original Windows driver for that.
But I doubt it will fix your problems.
You should try to use a different kernel or distribution. You could use some life-CD or something that doesn't need to be installed to check this.

magvay, 2016-07-09:
magvay@Dell-L502X:~$ lsusb -vv

Bus 002 Device 004: ID 8086:0189 Intel Corp. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          224 Wireless
  bDeviceSubClass         1 Radio Frequency
  bDeviceProtocol         1 Bluetooth
  bMaxPacketSize0        64
  idVendor           0x8086 Intel Corp.
  idProduct          0x0189 
  bcdDevice           69.19
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          177
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       224 Wireless
      bInterfaceSubClass      1 Radio Frequency
      bInterfaceProtocol      1 Bluetooth
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
[... 1499 lines cut ...]
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
magvay@Dell-L502X:~$ 

i have secondary OS on laptop: windows 10
but thinks new firmware not change sn.
and may be kill mouse ))) not legal sn = may be have not same and legal hardware inside mouse

mbuesch, 2016-07-15:
Fixed in razercfg-0.35.
Thanks for your bug report!

closed 2016-07-15 by mbuesch