mbuesch/razer #80: Support for Mamba Tournament Edition

haoanw, 2016-07-15:
Can somebody implement the support for Mamba TE or simply give me some hint how to do it, i try to implemented but it's really beyond my coding skills. Here i attach the data captured for wireshark just in case. Thx!

mamba.pcap (https://www.dropbox.com/s/dqxdg29uy6kl618/mamba_TE.pcap?dl=0)

haoanw, 2016-07-23:
Hi, I tried to implement the driver to the library following the example of Deathadder Chroma (hw_deathadder_chroma.h / hw_deathadder_chroma.c). Now it is successfully detected by the razercfg and qrazecfg(according to the usb id). And the DPI mapping, DPI mapping for individual axis both are working(however the DIP adjusting buttons are not working anymore, it used to work without the driver installed, i assume it needs its own mapping). 

Now I feel very difficult to find the lights information from the pcap file i recorded.  Here i attached both pcap file i recorded for Mamba TE and DA Chroma(this device is working properly). Can anyone or the original author of DA Chroma give me some hints about how to find the address of the lights or the defination of the commands for the lights. I would be happy to continue testing the Mamba TE, i feel it is almost there :)

Btw, during the recording of the pcap file, i triggered on all the possible lights and buttons for both devices.
[pcap files for both device.zip](https://github.com/mbuesch/razer/files/379459/pcap.files.for.both.device.zip)

haoanw, 2016-08-02:
Hi! After several days, I finally get it working (partially). Here i make a simple explanation and see if there is a need to commit.
1. I use the template of 'hw_deathadder_chroma.c' so there is no DPI adjusting buttons mapping for now, and i don's really need it so i leave them un-functional. Other buttons all work and DPI adjusting works with razercfg and qrazercfg.
2. The lighting control mechanism for mamba TE is quite different from the previous released devices, it got 16 individual lights on it, so there are no specific ID for each light, and they are controlled real-time with usb data at specific bits if you want to customized them individually.
3. Fortunately, this device has 5 default effects (modes) that only need a one-time instruction. they are: static, breathing, spectrum, wave and reaction. and the color can be edited as well. but each effect will take place for all the lights on it. 
So far i have achieved to switch among those 5 light modes and assign color, turn them on/off from both command-line and Qt based GUI. I am satisfied with this already so i won't do any further development. Anyone who need it can contact me or, if @mbuesch thinks it is okay to commit, then i would happy to do it too.

mbuesch, 2016-08-02:
Yes, show us some code please. I'll comment on actual code only. ;)

haoanw, 2016-08-02:
there you go, it's based on 'hw_deathadder_chroma.c' as i said. and also some slight changes in related files as well.

[hw_mamba_tournament_edition.zip](https://github.com/mbuesch/razer/files/397816/hw_mamba_tournament_edition.zip)

mbuesch, 2016-08-02:
This looks ok to begin with.
It seems to require some further changes in librazer and possibly razerd.
Can you please set up a pull request with everything in it that is needed to get it going?
Also, pleeeeeeeaaaaase fix the line lengths. Almost 500 characters per line is a _bit_ of a stretch. :) Try to keep line lengths low. Below 80 characters, 100 chars in exceptional cases at most.
Also the whole MAMBA_TOURNAMENT_EDITION prefix is a little long. Please just call it MAMBATE or something like that. Same goes for lower case variants.
In the lower LED functions there's some 0x06 constant which I presume is the LED-static value. Please use the enum constant there.

mbuesch, 2016-08-03:
I pulled this into master.

Please test it with my additional changes applied.
Thanks.

haoanw, 2016-08-03:
okay, i just tested, it works. thanks for formatting up the code, it looks much better now :)

mbuesch, 2016-08-04:
I released razercfg-0.36 with this feature included.
https://bues.ch/h/razercfg
Thanks for your support.

closed 2016-08-04 by mbuesch