mbuesch/razer #81: Razer Naga Chroma support

BenasPaulikas, 2016-07-30:
Hey thanks for this razer integration on linux but my mouse isn't detected :(.

benas@benas-X555LB ~ $ razercfg -L
No Razer device found in the system

benas@benas-X555LB ~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 004: ID 0e8d:763f MediaTek Inc. 
Bus 001 Device 003: ID 0bda:57b5 Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 1532:0053 Razer USA, Ltd 
Bus 001 Device 006: ID 22b8:2e82 Motorola PCS 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

benas@benas-X555LB ~ $ uname -a
Linux benas-X555LB 4.4.0-21-generic #37-Ubuntu SMP Mon Apr 18 18:33:37 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

Many thanks

mbuesch, 2016-07-30:
This device is not supported, yet.

Agronis, 2016-08-28:
+1 For hopeful request to be fulfilled.

justinnichols, 2016-09-08:
+1 From me as well.

chibbardpm, 2016-10-23:
+1 as well, fwiw, Microcenter doesn't shelve the non-Chroma razer anymore.

pcx436, 2016-12-02:
+1 From me too. Thank you for this creation. My old Razer mouse works fine and I can wait until the new one does too!

brianfreytag, 2017-12-26:
+1 From me as well. We're a year in here and nobody has gotten it to work?