mbuesch/razer #84: Razer Copperhead not being recoginzed anymore

mihab, 2016-08-07:
For some reason my Razer Copperhead is not being detected on latest 64-bit version. Running the scan and nothing shows up. Did anything change in the latest kernel/ubuntu version and is there any way do debug this? I also don't have xorg.conf on my system

mbuesch, 2016-08-07:
Which version of razercfg are you using?
What is the USB ID of your mouse?
Can you please run razerd with -l 3 and show the output?

The kernel should not have any effect on razercfg.

mihab, 2016-08-07:
Hm seems like something is stuck here:

miha@miha-GS70-2QE:~$ razercfg -v
Razer device configuration tool
Version 0.36
miha@miha-GS70-2QE:~$ xinput --list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Razer Razer Copperhead Laser Mouse      	id=15	[slave  pointer  (2)]
⎜   ↳ ETPS/2 Elantech Touchpad                	id=11	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Power Button                            	id=9	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=10	[slave  keyboard (3)]
    ↳ MSI WMI hotkeys                         	id=12	[slave  keyboard (3)]
    ↳ BisonCam, NB Pro                        	id=13	[slave  keyboard (3)]
    ↳ Razer Razer Copperhead Laser Mouse      	id=14	[slave  keyboard (3)]
miha@miha-GS70-2QE:~$ sudo systemctl stop razerd
miha@miha-GS70-2QE:~$ sudo razerd -l 3
Razer device service daemon
librazer: hw_copperhead: Got invalid profile nr in profile config
librazer: hw_copperhead: Failed to read config from hardware
^CTerminating razerd.    
miha@miha-GS70-2QE:~$ sudo systemctl start razerd
miha@miha-GS70-2QE:~$ 

Any clue what this error means?

mbuesch, 2016-08-07:
razercfg does not like what it reads from the hardware.
Did you use the windows driver in the mean time?

mihab, 2016-08-07:
Yes I dual boot all the time, was thinking maybe windows messed it up but I do not use the razer tools in windows. Is there any way to reset it the config?

mbuesch, 2016-08-07:
I think maybe the Windows driver set the device to an invalid state. razercfg is very strict about that.
I'll mark this as a bug, although I don't see a bug in razercfg, yet. But razercfg could be less strict in the error checks to workaround this issue.

Can you try this please:
https://github.com/mbuesch/razer/commit/9ffa3c1f9c295bfd035504cf32cc12e80be72159
(branch tmp-testing2)

mbuesch, 2016-08-07:
Whoops, this was incomplete. Give me a second. :)

mihab, 2016-08-07:
no rush i'm using the ppa atm so will try to compile the branch tomorrow when you say it's ready. Thanks a lot for checking

mbuesch, 2016-08-07:
Please test this:
https://github.com/mbuesch/razer/commit/b78c0482e78935686f1a95c35842fa983431d849
(branch tmp-testing2)

mihab, 2016-08-08:
ok thanks will do when I get home later today

mihab, 2016-08-09:
I can confirm the branch works fine, this is the output I get:

miha@miha-GS70-2QE:~/Dev/razer$ sudo razerd -l 3
Razer device service daemon
librazer: hw_copperhead: Got invalid profile nr in profile config: 5
librazer: hw_copperhead: Got invalid profile nr in profile config: 4
[razerd debug]: librazer: Applying config section "Mouse:Copperhead*:*:*" to "Mouse:Copperhead:USB-003-003:1532-0101-0"
[razerd debug]: librazer: Allocated and initialized new mouse "Mouse:Copperhead:USB-003-003:1532-0101-0"
[razerd debug]: Broadcasting mouse-add event
[razerd debug]: Client connected (fd=11)
[razerd debug]: Client disconnected (fd=11)
[razerd debug]: Client connected (fd=12)
[razerd debug]: Client disconnected (fd=12)
^CTerminating razerd.
[razerd debug]: librazer: Freeing mouse (type=3)
[razerd debug]: Broadcasting mouse-remove event
miha@miha-GS70-2QE:~/Dev/razer$ sudo systemctl start razerd

I can see and configure the mouse in qrazercfg now. Would be nice to get this merged to master. Thanks a lot for the fix!

mbuesch, 2016-08-09:
Thanks for your help.
This will be part of razercfg-0.37.

mihab, 2016-08-09:
Fantastic thanks a lot!

mbuesch, 2016-08-10:
Fixed in razercfg-0.37
https://bues.ch/h/razercfg

closed 2016-08-10 by mbuesch