mbuesch/razer #85: Razer Naga Hex V2

l3iggs, 2016-08-17:
@mbuesch 
Hi, I just bought [one of these](http://www.razerzone.com/gaming-mice/razer-naga-hex-v2). Looks like it's not supported here. Are you aware of its existence? I'd like to try to add support for it. Would you estimate this would be a lot of work? trivial?

`lsusb -v` gives
```
Bus 001 Device 010: ID 1532:0050 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0050 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Naga Hex V2
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

mbuesch, 2016-08-17:
The Hex v2 is a known unsupported device. (That's why it says v1 on the homepage ;) )

It is not possible to estimate the effort. That can go from just adding a USB ID to writing a completely new hardware driver module.
A mouse having a predecessor even with the same name does not necessarily tell anything about its API.

But feel free to work on support for this device.

l3iggs, 2016-08-17:
I've attempted to attach here a capture I've just made of me using the Windows Razer mouse config tool to map the button marked `1` on the mouse to the `1` key on my keyboard.

@mbuesch does it look like I've captured something useful here?

[button1-mapped-to-kb-1.zip](https://github.com/mbuesch/razer/files/423556/button1-mapped-to-kb-1.zip)

mbuesch, 2016-08-17:
Well, it looks like captured control data.
Please compare that to the existing naga-hex driver (and possibly others).
At the top of the C file you can see the on-wire data structures (those with "packed" attribute).

l3iggs, 2016-08-17:
Well, [`struct naga_command`](https://github.com/mbuesch/razer/blob/master/librazer/hw_naga.c#L50) is 90 bytes long and my capture has a few packets with a 90 byte long field that wireshark calls "CONTROL response data". The crc in the struct seems to line up with a crc looking byte in my captures too.

mbuesch, 2016-08-17:
Good start.
Please check if the structures for all commands (LEDs, DPIs, etc...) match (or not).

l3iggs, 2016-08-17:
I've just installed a build from [my fork](https://github.com/l3iggs/razer).  
The mouse shows up in the dropdown in qrazercfg now, but that's about all that works.
Toggling "Glowing Logo LED" makes the cursor disappear :fearful: and puts a bunch of bad things in my system log:
```
Aug 17 22:39:11  /usr/lib/gdm/gdm-x-session[1693]: (II) config/udev: removing device Razer Razer Naga Hex V2
Aug 17 22:39:11  /usr/lib/gdm/gdm-x-session[1693]: (**) Option "fd" "38"
Aug 17 22:39:11  /usr/lib/gdm/gdm-x-session[1693]: (II) UnloadModule: "libinput"
Aug 17 22:39:11  /usr/lib/gdm/gdm-x-session[1693]: (II) systemd-logind: releasing fd for 13:77
Aug 17 22:39:11  razerd[1352]: librazer: razer-naga: Command 0003/0401 failed with 03
...
```
The firmware version shows 1.00 (don't know what it should be).
So it looks like I've got more work to do.

l3iggs, 2016-08-20:
I'm going to put my captures here for reference. Let me know if I should put them elsewhere.

l3iggs, 2016-08-20:
map keyboard key "2" to side mouse button 2
[button2-mapped-to-kb-2.zip](https://github.com/mbuesch/razer/files/428277/button2-mapped-to-kb-2.zip)

l3iggs, 2016-08-20:
set acceleration to 1
[set-accel-to-1.zip](https://github.com/mbuesch/razer/files/428278/set-accel-to-1.zip)

l3iggs, 2016-08-20:
set acceleration to 10
[set-accel-to-10.zip](https://github.com/mbuesch/razer/files/428279/set-accel-to-10.zip)

l3iggs, 2016-08-20:
set dpi to 100
[set-DPI-to-100.zip](https://github.com/mbuesch/razer/files/428280/set-DPI-to-100.zip)

l3iggs, 2016-08-20:
set dpi to 16000
[set-DPI-to-16000.zip](https://github.com/mbuesch/razer/files/428281/set-DPI-to-16000.zip)

l3iggs, 2016-08-20:
set polling rate to 125 Hz
[set-polling-rate-to-125.zip](https://github.com/mbuesch/razer/files/428283/set-polling-rate-to-125.zip)

l3iggs, 2016-08-20:
set polling rate to 1000 Hz
[set-polling-rate-to-1000.zip](https://github.com/mbuesch/razer/files/428284/set-polling-rate-to-1000.zip)

l3iggs, 2016-08-20:
set lighting effect to none
[set-lighting-effect-none.zip](https://github.com/mbuesch/razer/files/428287/set-lighting-effect-none.zip)

l3iggs, 2016-08-20:
set lighting effect to static
[set-lighting-effect-static.zip](https://github.com/mbuesch/razer/files/428289/set-lighting-effect-static.zip)

l3iggs, 2016-08-20:
change static lighting to blue
[set-lighting-effect-staticBlue.zip](https://github.com/mbuesch/razer/files/428290/set-lighting-effect-staticBlue.zip)

l3iggs, 2016-08-20:
change static lighting to red
[set-lighting-effect-staticRed.zip](https://github.com/mbuesch/razer/files/428291/set-lighting-effect-staticRed.zip)

l3iggs, 2016-08-20:
change static lighting to green
[set-lighting-effect-staticGreen.zip](https://github.com/mbuesch/razer/files/428292/set-lighting-effect-staticGreen.zip)

l3iggs, 2016-08-20:
set lighting brightness to 33%
[set-lighting-brightness-33.zip](https://github.com/mbuesch/razer/files/428293/set-lighting-brightness-33.zip)

l3iggs, 2016-08-20:
set lighting brightness to 100%
[set-lighting-brightness-100.zip](https://github.com/mbuesch/razer/files/428294/set-lighting-brightness-100.zip)

l3iggs, 2016-08-20:
set lighting effect to spectrum cycling
[set-lighting-effect-spectrumCycling.zip](https://github.com/mbuesch/razer/files/428296/set-lighting-effect-spectrumCycling.zip)

phamhuudanh96, 2017-01-16:
I just bought one and really want to help! Is there any quick guide for me?

z3ntu, 2017-01-16:
I think only this document exists for this project: https://github.com/mbuesch/razer/blob/master/HACKING.md

mbuesch, 2017-01-16:
In addition to that the information from the website:
https://bues.ch/h/razercfg#how_to_add_support_for_new_devices

l3iggs, 2017-01-18:
Quick update from my end--
I haven't really made any further progress. I stopped when I realized that adding support for this mouse isn't as trivial as adding the USB ID and re-using one of the already existing drivers. Unfortunately, I just don't have the time to get this working. I've dropped my captures here in hopes that the next dev that comes along with time to work on this might find them useful.

z3ntu, 2017-01-18:
For anyone else, it's supported by https://github.com/terrycain/razer-drivers

l3iggs, 2017-01-18:
:astonished: