mbuesch/razer #86: Razer DeathAdder 3500 DPI is not detected at all

ddnomad, 2016-09-29:
```
ddnomad@dracula:~$ xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Razer Razer DeathAdder                  	id=11	[slave  pointer  (2)]
⎜   ↳ ETPS/2 Elantech Touchpad                	id=15	[slave  pointer  (2)]
```

razercfg doesn't detect my mouse despite the fact it's listed and detected correctly by X. 

```
ddnomad@dracula:~$ razercfg -d mouse
No Razer mouse found in the system
```

I even built the latest snapshot from this repo but the issue still exists. 
If I can help you somehow - just say how :)

Some more system info: (lsinput output)
```
/dev/input/event7
   bustype : BUS_USB
   vendor  : 0x1532
   product : 0x54
   version : 273
   name    : "Razer Razer DeathAdder"
   phys    : "usb-0000:00:14.0-1/input0"
   uniq    : ""
   bits ev : EV_SYN EV_KEY EV_REL EV_MSC
```

mbuesch, 2016-10-16:
Are you sure this really is a DeathAdder 3500 DPI? If so, Razer seems to have created a new device with an old name.

ddnomad, 2016-10-18:
Well, as it clearly stated at the bottom of a mouse it should be DeathAdder 3500 DPI. See a picture: https://imgur.com/gallery/PLwxB

h0rn3t, 2016-11-15:
I also have that model :(((((((

h0rn3t, 2016-11-15:
@mbuesch  what I need to do? run wireshark on windows and sniff usb, is it right?

mbuesch, 2016-11-15:
https://bues.ch/h/razercfg#how_to_add_support_for_new_devices

h0rn3t, 2016-11-15:
https://dl.dropboxusercontent.com/u/32082774/razer3500dpi.pcapng

h0rn3t, 2016-11-15:
@ddnomad you can use https://github.com/mbuesch/razer/pull/94 I tested it, all functions working fine

ddnomad, 2016-11-16:
Thank you a lot!

I'll check it out.

h0rn3t, 2016-11-16:
@ddnomad  for now you can take this more adapted version  https://github.com/h0rn3t/razer/releases/tag/razercfg-0.38.1
now I working on it