mbuesch/razer #92: DeathAdder Elite (2016)

lswest, 2016-11-12:
Is there anyone working on support for this device yet?

Based off the functionality, I'm not sure it's that much different from the chroma version - most of the changes were to the shape, the sensor, and the buttons themselves.  The USB ID is: `1532:005c`.

mbuesch, 2016-11-12:
This device is not supported and as far as I know nobody is working on it, yet.

radical-equanimity, 2017-02-25:
+1 for this issue request

Pillgar, 2017-03-09:
Would love to see this mouse supported.  Much better tracking with the new optical sensor.

radical-equanimity, 2017-03-10:
I started to develop this by assuming it operates the same as Deathadder Chroma, just with added enum values for DPI stages. Turns out this is not the case. The device is recognized but the GUI doesn't seem to issue the correct commands to the device. I captured USB traffic for certain LED-changing scenarios but it's not easy to reverse the protocol from this subset of captures. Needs more investigation

fuzunspm, 2020-03-06:
I want to contribute to add support for Elite. I also have the hardware

mbuesch, 2020-03-06:
Great. Please don't hesitate to do so.

skewty, 2020-04-17:
@lswest perhaps embed the USB ID in the issue subject?  My Elite is `1532:005c` I don't know what (2016) means or if that will help others.

```shell
$ lsusb | grep Razer
Bus 002 Device 007: ID 1532:005c Razer USA, Ltd 
$
```