mbuesch/razer #93: How Do I open this?

INeedHelpNoob, 2016-11-12:
I'm using ubuntu 16.04 and I can't seem to open it. I have installed and done all the instructions with no errors just dont know how to open it. Sorry for being a noob

mbuesch, 2016-11-12:
There basically are three applications:

qrazercfg - The graphical user interface
qrazercfg-applet - The graphical tray applet
razercfg - The command line interface

Just run these just like any other application on your operating system.

INeedHelpNoob, 2016-11-12:
Is there any commands I can do to set the dpi?

INeedHelpNoob, 2016-11-12:
I fixed the issue and discovered why it wasn't working but thanks for the help and cheers for the program

mbuesch, 2016-11-12:
if you run qrazercfg a graphical dialog box will open.
It will look similar to this https://bues.ch/images/qrazercfg.png
The exact layout depends on your hardware.

mbuesch, 2016-11-12:
>I fixed the issue and discovered why it wasn't working but thanks for the help and cheers for the program

Have fun! :)

closed 2016-11-12 by mbuesch