mbuesch/razer #97: macOS/OSX Port

ed1d1a8d, 2017-02-02:
I'm looking to port some of the drivers (specifically ones neeeded for deathadd2013) onto macOS Sierra. I built and installed using the linux instructions. But when running the daemon, I got the error:

`Unable to claim USB interface (Permission denied)`

From what Googling I could do, this seems to be an issue with system kexts recognizing the device and taking control away from razerd. I was wondering what the best solution to this would be -- somehow get ourselves an unsigned kext or disable system kexts (this might break other things)?

One idea I had was to perhaps write a version of razerd using [HIDAPI](http://www.signal11.us/oss/hidapi/) as the backend instead of libusb. Quoting the [libusb wiki](http://www.libusb.org/wiki/APIs):

> While it is possible to use libusb with HID class devices it requires the original HID class driver to be replaced by a driver which supports libusb, and this leads to a quite poor user experience on any platform other than Linux. It is recommended to use ​HIDAPI instead. HIDAPI provides convenient cross-platform access to HID class devices and can do so without any kernel driver concerns across all supported platforms.

How feasible would this be? I am willing to do this myself, but would need some guidance on what exactly libusb is doing right now and where it is used.

z3ntu, 2017-02-02:
Doesn't macOS have the official Razer Synapse?

ed1d1a8d, 2017-02-02:
Yes, but I was hoping to just use a driver without the whole cloud connection thing.

mbuesch, 2017-02-02:
>But when running the daemon, I got the error:
>Unable to claim USB interface (Permission denied)

Are you trying to run the razerd as root?

ed1d1a8d, 2017-02-02:
Yes I ran `sudo razerd`.