mbuesch/razer #100: Razer Naga Epic dock support

etxzay, 2017-02-15:
It works when mouse connected wired, but when it's connected via dock and bluetooth - device not found.

```
etxzay@blink ~> uname -a
Linux blink 4.4.0-62-generic #83-Ubuntu SMP Wed Jan 18 14:10:15 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

```
```
etxzay@blink ~> lsusb | grep Razer
Bus 003 Device 003: ID 1532:0021 Razer USA, Ltd

```
```
etxzay@blink ~> razercfg -s
etxzay@blink ~>
```

mbuesch, 2017-02-15:
Thanks for your message.
Unfortunately razercfg is not a USB device driver for your device. It's a userspace program to configure razer mice. The actual HID driver for your device is (supposed to be) located in the kernel. So this is a kernel issue.

So I will close this issue as wont-fix, because we can't to anything about this in razercfg.
Please contact the kernel HID developers.

closed 2017-02-15 by mbuesch

etxzay, 2017-02-15:
@mbuesch so I plugged in my mouse, configured it with your "userspace program". Disconnected a wire, plugged in like a bluetooth with razer docked (complected with a mouse) and all configuration goes to hell. Nice configuring buddy xD

mbuesch, 2017-02-15:
@etxzay 
I don't really understand what you are saying and I'm not even sure I understood your original post correctly.
If you want support, please describe in detail what you do and what goes wrong.
Thanks for your help.

etxzay, 2017-02-15:
@mbuesch probably you better go and learn what razer naga epic is. It's a bluetooth mouse packaged with dock station for charge, bluetooth connection and a single cord. You have 2 ways to organize mice connection to PC. First is with cord connected directly to the mice, in this case your "userspace program" seems working. Second is connecting a cord to the dock and mice via bluetooth to the dock. In this case software looks useless cause it can not find any device. And it does not matter how you configured you mice in wired connection.

mbuesch, 2017-02-15:
@etxzay 
>probably you better go and learn what razer naga epic is.

I'd like to keep the discussion on a respectful level. Thanks.

>First is with cord connected directly to the mice, in this case your "userspace program" seems working.

Ok. Very nice indeed. I don't understand why you put userspace program into quotes. But anyway. Nice.

>In this case software looks useless cause it can not find any device.

So and this is the critical point you need to explain in detail.
Which part of the system does not find any device? Is it not found by the operating system, or does it just not show up in razercfg but work fine otherwise.

> And it does not matter how you configured you mice in wired connection.

What does that mean? Please explain this.

etxzay, 2017-02-16:
> I'd like to keep the discussion on a respectful level. Thanks.

Respectful level is when you dive a little into a problem before just closing the issue.

> Which part of the system does not find any device? Is it not found by the operating system, or does it just not show up in razercfg but work fine otherwise.

The answer in my first post in this issue. Naga connected via razer bluetooth dock, it works, system found it and recognized as a razer mice. Your software does not properly works with it. 

P.S.: You better remove "Stable" status for "Razer Naga Epic mouse 1532:001F" from your site and it's over. I already removed this stuff from my PC, so I don't mind.

mbuesch, 2017-02-16:
> Respectful level is when you dive a little into a problem before just closing the issue.

Well, there obviously is some kind of communication problem happening here.
I closed the issue, because it seemed like this problem was not related to razercfg.

If you think it is related to razercfg, you need to explain the problem in detail to me.

> The answer is in my first post in this issue.

No, it's not. The first post basically says "mouse does not work". And that is not the job of razercfg to make your mouse work.
But obviously that is not what you meant to say.

> Naga connected via razer bluetooth dock, it works, system found it and recognized as a razer mice. Your software does not properly works with it.

Ok. But please provide more details on this. Otherwise I'm not able to help.
What is the USB ID of the device? Does it change when being docked vs being not docked?
Please provide all related logs that your system has to offer and clearly state whether the mouse was in docked or undocked state at the time of the message.

> P.S.: Please, remove "Stable" status for "Razer Naga Epic mouse 1532:001F" from your site and it's over.

No. I don't see a reason to do so. The driver works well with the device 1532:001F and as of now there is no indication this status has to change.

> I already removed this stuff from my PC, so I don't mind.

If you are still interested in cooperating, please re-open this issue and provide detailed information.

Please keep in mind that this is community based free software and not the support channel of some commercial company. We do the best we can to support lots of razer mice. But that does not mean there aren't any problems.
Thanks for your time.

z3ntu, 2017-02-16:
If I may add something:
`1532:0021` is the dock which is connected via an USB cable to the computer. This dock has additional lighting capabilities.
If I might split @etxzay s first message up:
> It works when mouse connected wired

Understandable.
> , but when it's connected via dock and bluetooth - device not found.

`razercfg` outputs no device when the mouse isn't connected wired but via the dock / via bluetooth (see further down). I don't have the mouse myself so I can only write what I 'know'. How exactly the whole connection thing, so I don't know if 
```
computer --bluetooth--> mouse
                --usb--> dock
```
or 
```
computer --usb--> dock --bluetooth--> mouse
```
is correct.

And as far as I can understand it, the mouse works well via bluetooth but razercfg doesn't see the mouse / doesn't have a driver for that USB PID.

mbuesch, 2017-02-16:
Oh finally some background information. Thanks a lot @z3ntu 

>And as far as I can understand it, the mouse works well via bluetooth but razercfg doesn't see the mouse / doesn't have a driver for that USB PID.

Ok. Question to @etxzay or anybody else who knows:
Does the mouse appear as an ordinary USB HID device to the computer, if connected wirelessly?
Or is the USB device that shows up just a plain BT transceiver? (Like any other BT dongle)