mbuesch/razer #102: ModuleNotFoundError

sum01, 2017-06-27:
Installed via the [AUR](https://aur.archlinux.org/packages/razercfg/), first time trying to run it (I have a Razer Naga 2014). 
Version/pkgrel `0.39-1` when downloaded.
Linux version `4.11.6-3-ARCH`
```
cd /usr/bin/
./qrazercfg

Traceback (most recent call last):
  File "./qrazercfg", line 19, in <module>
    from PySide.QtCore import *
ModuleNotFoundError: No module named 'PySide'
```
Referenced line error https://github.com/mbuesch/razer/blob/master/ui/qrazercfg#L19

I installed `python2-pyside` (`1.2.4-6`) and `python2-pyside-tools` (`0.2.15-4`) from the official repo, thinking that was the issue, but I still get the same error even after a quick uninstall/reinstall.

sum01, 2017-06-28:
Switched to `python-pyside` and now the error is different... 
`Failed to connect to razerd socket: [Errno 2] No such file or directory`

polyzen, 2017-06-28:
https://github.com/mbuesch/razer#if-you-use-systemd
https://github.com/mbuesch/razer#using-the-tools

sum01, 2017-06-28:
Ha, that's my bad. Guess I should've read more.

closed 2017-06-28 by sum01