mbuesch/razer #104: Compile for python 2.7?

dmitry-saritasa, 2017-07-19:
I have both python 2.7 and 3.5 installed in my Fedora 25

```
➜  razer git:(master) ✗ razercfg           
Traceback (most recent call last):
  File "/usr/local/bin/razercfg", line 22, in <module>
    from pyrazer import *
ImportError: No module named 'pyrazer'
➜  razer git:(master) ✗ razercfg -s
Traceback (most recent call last):
  File "/usr/local/bin/razercfg", line 22, in <module>
    from pyrazer import *
ImportError: No module named 'pyrazer'
➜  razer git:(master) ✗ echo $PYTHONPATH           

➜  razer git:(master) ✗ sudo find /usr -iname pyrazer                                                             
/usr/local/lib/python3.5/site-packages/pyrazer
➜  razer git:(master) ✗ cmake . 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dmitry/Development/tools/razer
➜  razer git:(master) ✗ make        
[ 87%] Built target razer
[ 95%] Built target razerd
[100%] Built target pyrazer
➜  razer git:(master) ✗ sudo make install            
[ 87%] Built target razer
[ 95%] Built target razerd
[100%] Built target pyrazer
Install the project...
-- Install configuration: ""
-- Up-to-date: /usr/lib/udev/rules.d/80-razer.rules
-- Up-to-date: /etc/pm/sleep.d/50-razer
-- Up-to-date: /usr/lib/systemd/system/razerd.service
-- Up-to-date: /usr/local/share/applications/razercfg.desktop
-- Up-to-date: /usr/local/share/icons/hicolor/scalable/apps/razercfg.svg
-- Up-to-date: /usr/local/share/icons/hicolor/scalable/apps/razercfg-alt.svg
-- Up-to-date: /usr/local/share/icons/hicolor/scalable/apps/razercfg-snake.svg
-- Up-to-date: /usr/local/lib/librazer.so.1
-- Up-to-date: /usr/local/lib/librazer.so
-- ldconfig: 0
-- Up-to-date: /usr/local/bin/razerd
-- systemctl --system daemon-reload: 0
-- systemctl --system --force enable razerd.service: 0
-- udevadm control --reload-rules: 0
running install
running build
running build_py
running build_scripts
running install_lib
running install_scripts
changing mode of /usr/local/bin/qrazercfg-applet to 775
changing mode of /usr/local/bin/razercfg to 775
changing mode of /usr/local/bin/qrazercfg to 775
running install_egg_info
Removing /usr/local/lib/python3.5/site-packages/razercfg-0.39-py3.5.egg-info
Writing /usr/local/lib/python3.5/site-packages/razercfg-0.39-py3.5.egg-info
-- Up-to-date: /usr/local/bin/razer-gamewrapper

```

It can't run under python 3.5 as it doesn't install PySide that is needed for the utility to work. So how do I install it into **/usr/lib/python2.7/site-packages/**

mbuesch, 2017-07-19:
razercfg does not support Python 2.7.
So you need to install PySide for Python 3.x to get it running.
Please check your distribution package archive.

dmitry-saritasa, 2017-07-19:
okay I see, now I get the following error

```
➜  ~ razercfg -s            
Failed to connect to razerd socket: [Errno 2] No such file or directory
➜  ~ systemctl status razerd
● razerd.service - Razer device state daemon
   Loaded: loaded (/usr/lib/systemd/system/razerd.service; enabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since Wed 2017-07-19 10:08:36 PDT; 4min 13s ago
  Process: 10331 ExecStart=/usr/local/bin/razerd -f (code=exited, status=127)
 Main PID: 10331 (code=exited, status=127)

➜  ~ sudo find /usr -iname 'libraz*'
/usr/lib/librazer.so
/usr/lib/librazer.so.1

```

dmitry-saritasa, 2017-07-19:
okay I figured out the problem with .so libs, but now

```
➜  ~ razercfg   
Razer device configuration tool
Version 0.39

Usage: razercfg [OPTIONS] [-d DEV DEVOPS] [-d DEV DEVOPS]...

-h|--help            Print this help text
-v|--version         Print the program version number
-B|--background      Fork into the background
-s|--scan            Scan for devices and print the bus IDs
-K|--reconfigure     Force-reconfigure all detected devices

-d|--device DEV      Selects the device with the bus ID "DEV"
    Use the special value "mouse" for DEV to select
    the first razer mouse device found in the system.
    If this option is omitted, the first Razer device found is selected.

-S|--sleep SECS      Sleep SECS seconds.

Device operations (DEVOPS):
These options apply to the device that is specified with -d

Options for mice:
-V|--fwver                          Print the firmware version number
-p|--profile PROF                   Changes the active profile
-P|--getprofile                     Prints the active profile
-r|--res [PROF:]RES[xRES]           Changes the scan resolution
-R|--getres                         Prints the resolutions
-f|--freq [PROF:]FREQ               Changes the scan frequency
-F|--getfreq                        Prints the frequencies
-L|--leds                           List the device supported LED identifiers
-l|--setled [PROF:]LED:(on|off)     Toggle the LED with the identifier "LED"
                                    Use the special identifier "all"
                                    to toggle all supported LEDs.
-c|--setledcolor [PROF:]LED:rrggbb  Set LED color to RGB 'rrggbb'
-m|--setledmode  [PROF:]LED:MODE    Set LED mode to MODE ('static', 'spectrum'
                                                               or 'breathing')

-X|--flashfw FILE                   Flash a firmware image to the device

The profile number "PROF" may be 0 for the current profile. If omitted,
the global settings are changed (not possible for every device).
➜  ~ razercfg -s
➜  ~ lsusb           
Bus 002 Device 002: ID 8087:8002 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:800a Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 045b:0210 Hitachi, Ltd 
Bus 004 Device 002: ID 05e3:0612 Genesys Logic, Inc. Hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 011: ID 047d:8018 Kensington 
Bus 003 Device 013: ID 041e:4089 Creative Technology, Ltd 
Bus 003 Device 012: ID 041e:4091 Creative Technology, Ltd 
Bus 003 Device 009: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 007: ID 1bcf:053a Sunplus Innovation Technology Inc. Targa Silvercrest OMC807-C optische Funkmaus
Bus 003 Device 005: ID 045b:0209 Hitachi, Ltd 
Bus 003 Device 010: ID 1e54:2030 TypeMatrix 2030 USB Keyboard
Bus 003 Device 008: ID 1532:003f Razer USA, Ltd 
Bus 003 Device 006: ID 1038:160c SteelSeries ApS 
Bus 003 Device 003: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 004: ID 21b4:0230  
Bus 003 Device 002: ID 0424:2412 Standard Microsystems Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

➜  ~ lsusb G razer
15:Bus 003 Device 008: ID 1532:003f Razer USA, Ltd
```

so razercfg -s shows no devices, although I have Naga Epic mouse

mbuesch, 2017-07-19:
>15:Bus 003 Device 008: ID 1532:003f Razer USA, Ltd

That device is currently not supported.

dmitry-saritasa, 2017-07-19:
I see, clear. Thanks! Are there substantial changes between 001f and 003f?

mbuesch, 2017-07-19:
>Are there substantial changes between 001f and 003f?

I don't know. There might be only very tiny differences, or it might be a completely new device.

z3ntu, 2017-07-19:
`001f` is the Razer Naga Epic and `003f` the Razer Naga Epic Chroma