mbuesch/razer #115: Please upgrade to QT5

blshkv, 2018-03-23:
My distribution (Gentoo) has move away from QT4 and had to remove razer GUI support
Please migrate to QT5 as well.

mbuesch, 2018-03-24:
Patches or pull requests are welcome.

mbuesch, 2018-06-10:
I added PyQt5 support 
Can you please download the latest Razercfg snapshot from here, test it and give me your feedback?
https://bues.ch/gitsnap/razer

Thanks for your support.

blshkv, 2018-06-11:
The UI seems to be working. I have compiled and run it, no immidiate glitches. Feel free to release 0.40 at some stable moment.

Thanks!

blshkv, 2018-06-11:
https://github.com/mbuesch/razer/commit/598b8d1da086603993c203d33fc714c04780a92f

mbuesch, 2018-06-11:
Thanks for testing.
I'll re-open the issue until razercfg-0.40 will be released.

polyzen, 2018-10-14:
Applied the PyQt5 patch directly to 0.39, and everything's looking good.

blshkv, 2018-10-14:
@mbuesch  It's about time to cut a next release  ;-)

mbuesch, 2018-10-17:
Thanks for testing. I just released razercfg-0.40.

closed 2018-10-17 by mbuesch

polyzen, 2018-10-18:
For some reason the tray icon is not shown if you build 598b8d1 and on. It works just fine when the patch is applied directly to `0.39`. This is especially odd as the commits in between don't seem to touch anything relevant.

mbuesch, 2018-10-18:
@polyzen I currently assume that this is a Qt or PyQt bug. Which version of Qt and PyQt do you use?

blshkv, 2018-10-18:
I never used it but after installing the 0.40 release and running ```qrazercfg-applet``` manually it did show up.
```
dev-python/PyQt5-5.10.1-r1
dev-qt/qtcore-5.11.1-r1
```