mbuesch/razer #118: Binding keyboard buttons

Astaoth, 2018-04-23:
Hello,
Is there a way to assign a keyboard button to the mouse buttons ? If not, do you think it's possible to add this feature ?
BR

pushytoxin, 2018-06-09:
Check out https://wiki.archlinux.org/index.php/Mouse_buttons#Binding_keyboard_to_mouse_buttons

Although it isn't *entirely* out of scope for qrazercfg, adding and maintaining such a feature would be a lot of work.

Astaoth, 2018-06-10:
Actually, I think it is fully in the scope of qrazercfg since this are configuration options of a razer mouse. Can you, at least, add the possibility to bind on the mouse, the mouse button 6 7 8 and 9 ? Because currently I don't know how to map my mouse buttons in order to have them reconised by xbindkeys.
With xbindkeys, do you know how to manage few profiles ?

mbuesch, 2018-06-10:
Razercfg does only configure the Razer hardware. It is not supposed to do some configuration in the operating system or X11 or anywhere else.
The only thing razercfg does is send configuration commands to the hardware. And thus it can only provide options that the hardware is capable of.

If some remapping of HID button events is desired, that can never be done in razercfg, because razercfg does not see any HID events. This is what the kernel and X11 drivers are for.

Astaoth, 2018-06-10:
So can I expose the buttons in order to configure them with xbindkeys ? I can't leave them as dpi mapping or profile switch in this case. Should I configure them as "left click" or "windows button X" ?

mbuesch, 2018-08-08:
I'm sorry, I don't know.
Closing this issue here, because xbindkeys really is off topic here.
Please re-open it, if there is anything that needs to be changed in razercfg on the hardware level to support your needs.

closed 2018-08-08 by mbuesch