mbuesch/razer #123: Razer Lachesis 5600 DPI support

TheTharin, 2018-10-29:
Hello! I've got a problem with Lachesis 5600 DPI Mouse. I'd installed razercfg and saw that some of the buttons were mapped incorrectly. I tried remapping them and ended up breaking my mouse hardware mappings for side buttons and for the scroll-wheel. They all perform left-click now even in windows. Is there a way to change them back? I tried Razer Lachesis 5600 config tool on windows to no avail. It doesn't change them at all. Neither does razercfg. I see that this mouse is currently unsupported but can you at least give me a clue on how to fix my mouse? Thanks in advance!

mbuesch, 2018-10-29:
The wire protocol of that mouse is currently not fully reverse engineered and the driver is incomplete.
So I'm not really sure what happened and how to fix this.
If the mouse still responds to commands, I'd say this is fixable.
But that most likely means to fix the driver code first to talk correctly to the mouse.

The relevant parts are in synapse.c/h and buttonmapping.c/h

TheTharin, 2018-10-29:
Hello again! Oh, I have a spare mouse anyway, I can wait. Sorry, I don't really know C, so all I can is wait for you to ever get your hands on the problem patiently :D

mbuesch, 2018-11-08:
I will deactivate Lachesis 5600 support for now.
It is being marked as broken on the website, but to avoid breaking more devices I'll turn it off.

closed 2021-12-15 by TheTharin