mbuesch/razer #127: Quick question

Royalgamer06, 2018-11-08:
I have a Razer Deathadder 2013 and I hate the blinking light, which you can't set to static unfortunately. Before I embark into getting this github project to work, do you guys think it would be possible with this to make the LED constant (by e.g. constantly sending LED "on")?

mbuesch, 2018-11-08:
To the best of my knowledge the hardware doesn't support switching LED modes.
So it can only be switched on or off.
And that's what razercfg supports.

Does the Windows driver support changing LED modes?

Royalgamer06, 2018-11-08:
All I have in the razer synapse configurator are these options:
[image]

But yes, I also think it might be hardware related. However, switch from off to on, will always start the LED at it's brightest (and then it will start blinking). So I thought maybe having a script constantly telling it to turn LED on, will make it static.

mbuesch, 2018-11-08:
>So I thought maybe having a script constantly telling it to turn LED on, will make it static.

You could do such a thing by calling razercfg from a tiny script in a loop.

closed 2019-05-20 by mbuesch