mbuesch/razer #132: Lachesis Classic click delay problem

Nitrooo, 2019-11-25:
Today I was messing with `razercfg`, that has always worked very well, when my mouse started to behave badly. When I click the left or right buttons it becomes unresponsive for about one second (the cursor freezes) and the click is not registered. I have to hold the button for one second for the click to be registered. Same one second delay for the release. It seems like there's nothing I can do to make it work properly. I plugged the mouse to another computer running Windows and it's just the same. Resetting the mouse (by pressing the "profile button" for 10 seconds) didn't help. The sensitivity is reset, but the problem persists. It's completely unusable.

So should I throw my otherwise perfectly working mouse in the trash? Is there anything else that I could try before trashing it? I'd try flashing an old firmware but I can't find any information about the `razercfg -X` command. Can anybody help me?

mbuesch, 2019-11-27:
Thanks for your report.
You could try flashing new firmware with the official Razer driver for Windows.
razercfg currently does not support firmware flashing at all (the option does nothing).

Nitrooo, 2019-11-27:
Thank you for your answer. I figured that out later by looking at the source code.
The problem is that the latest firmware version I can find online is `1.94` while the version in my mouse is (reported as) `2.5755` and it won't accept an older version (the flashing process fails) . `razercfg -V` reports it as version `255.255` though.
My only hope is to take the `1.94` firmware and somehow change the version to something higher than `2.5755` with an hex editor but at first glance that's not an easy task.
I know this is OT, but if you have any idea, that would be appreciated.

mbuesch, 2019-11-28:
I'm sorry. I don't know of an easy solution to this problem.

mbuesch, 2020-06-08:
I'll close this issue.
Your problem may not have been solved, but I currently don't see how the community could provide further help.
Please feel free to re-open the issue, if you have further input about the issue.

closed 2020-06-08 by mbuesch