mbuesch/razer #133: Awake from sleep when mouse is moving

jichen-yang, 2020-05-09:
My operation system is Ubuntu 20.04 in Thinkpad L440 with DeathAdder 2013. When I move the mouse, the laptop can wake up from the sleep, whereas it does not happen for other mouses.

mbuesch, 2020-05-10:
Hi.
Thanks for your comment.
Unfortunately I don't think this has anything to do with this razercfg tool.
Please check the power management options in your operating system and your UEFI/BIOS.
You can probably disable the wakeup-by-usb somewhere along these settings.

mbuesch, 2020-06-08:
Did you find a solution to the problem?

jichen-yang, 2020-06-11:
> Did you find a solution to the problem?

I found a solution to the wakeup issue, [https://askubuntu.com/questions/509017/desktop-wakes-from-suspend-at-random-14-04](url)

closed 2020-06-12 by mbuesch