mbuesch/razer #137: item "res" invalid Mamba TE (Tournament Edition)

EXtremeExploit, 2021-08-12:
`systemctl status razerd`:
```
● razerd.service - Razer device state daemon
     Loaded: loaded (/usr/lib/systemd/system/razerd.service; enabled; vendor preset: disabled)
     Active: active (running) since Thu 2021-08-12 20:10:55 -03; 2min 28s ago
   Main PID: 2050542 (razerd)
      Tasks: 2 (limit: 19125)
     Memory: 396.0K
        CPU: 7ms
     CGroup: /system.slice/razerd.service
             └─2050542 /usr/bin/razerd -f

ago 12 20:10:55 pedro-arch systemd[1]: Started Razer device state daemon.
ago 12 20:10:56 pedro-arch razerd[2050542]: librazer: Config section "Mouse:Mamba*:*:*" item "res" invalid.
ago 12 20:10:56 pedro-arch razerd[2050542]: librazer: Config section "Mouse:Mamba*:*:*" item "led" invalid.
ago 12 20:10:56 pedro-arch razerd[2050542]: librazer: Failed to apply initial config to "Mouse:Mamba Tournament Edition:USB-003:1532-0046-PM1735137700304"
```

/etc/razer.conf:

```
[Mouse:Mamba*:*:*]
    disabled=false

    #DPI
    res=1:2000

    #Hz
    freq=1:1000

    #LEDs
    led=1:GlowingLogo:off
    led=1:ScrollWheel:off

    profile=1
```

Also been wondering how to set all leds to off in the conf file

mbuesch, 2021-08-13:
Hi. Thanks for your question.

The resolution of 2000 DPI is not supported by the device. Supported are:
```
        RAZER_MOUSE_RES_800DPI,
        RAZER_MOUSE_RES_1800DPI,
        RAZER_MOUSE_RES_3500DPI,
        RAZER_MOUSE_RES_5600DPI,
        RAZER_MOUSE_RES_10000DPI,
        RAZER_MOUSE_RES_16000DPI
```

The LED names "GlowingLogo" and "ScrollWheel" are also not available in the hardware. The name of the Mamba TE LED is "Basic".

EXtremeExploit, 2021-08-16:
so i guess ill have to stick with 1800 then, also Thanks for replying :D 
closing.

closed 2021-08-16 by EXtremeExploit