mbuesch/razer #138: SyntaxWarning: invalid escape sequence '\d'

KonradHoeffner, 2024-05-29:
I am using the Arch Linux AUR package <https://aur.archlinux.org/packages/razercfg> with Python 3.12.3 and razercfg prints this warning:

```
/usr/bin/razercfg:279: SyntaxWarning: invalid escape sequence '\d'
  m = re.match('^\d+$', arg)
/usr/bin/razercfg:283: SyntaxWarning: invalid escape sequence '\d'
  m = re.match('^(\d+)x(\d+)$', arg)
```

 Despite the warning, it still seems to work correctly.

mbuesch, 2024-05-29:
Thanks for your bug report.

Yes, a warning for invalid escape sequences has been added to Python.
This code will still work as expected, though. (unless they actually add valid a \d escape.)
I will fix it in the next version of razercfg.

mbuesch, 2024-05-29:
Do you want to check, if the latest `master` branch fixes the issue for you?
Thanks for your help.

KonradHoeffner, 2024-05-30:
Wow, thanks for the quick fix, it works perfectly!
I used the AUR package https://aur.archlinux.org/packages/razercfg-git which to my knowledge installs the newest version from the master branch and the warning is gone.

closed 2024-05-30 by KonradHoeffner