mbuesch/razer #139: Settings not saved after switching from Arch Linux AUR "razercfg" package to "razercfg-git"

KonradHoeffner, 2024-06-12:
On both my desktop PCs with Arch Linux and a Razer Deathadder Left, I used to install the [razercfg](https://aur.archlinux.org/packages/razercfg) package, however as that prints the warning below, I switched to [razercfg-git](https://aur.archlinux.org/packages/razercfg) which removes the warning, but now I have to configure the DPI every time on startup as "razercfg -r 3" is not remembered. I notice that /etc/razercfg.conf does not exist anymore.

I am not experienced enough with Arch Linux packages to know whether that is a problem that comes from this software or from the AUR package itself but in the latter case maybe I can help some other AUR users fix this problem.

My workaround right now is to copy /etc/razercfg.conf.pacsave to /etc/razercfg.conf.

```
/usr/bin/razercfg:279: SyntaxWarning: invalid escape sequence '\d'
  m = re.match('^\d+$', arg)
/usr/bin/razercfg:283: SyntaxWarning: invalid escape sequence '\d'
  m = re.match('^(\d+)x(\d+)$', arg)
Razer device configuration tool
Version 0.42
```

KonradHoeffner, 2024-06-13:
OK then this is not an issue with razercfg itself and I will close this issue, thanks for the help and the quick reply!

closed 2024-06-13 by KonradHoeffner

mbuesch, 2024-06-13:
I'm glad that helped. Have fun!